slic3r/Slic3r #2771: slicer 1.2.6 experimental (mac) vase gcode generation bad

maudette01432, 2015-04-04:
The preview of the layers in Slic3r's GUI show what appear to be correct walls but the Gcode generated jumps around and skips sections.  Use print settings Simple (bracelet), Filament: Taulman 618 Nylon Co-Polymer, Printer Simple Mode.

The resulting G-code ~line 228-232 jumps from  X-31.130 Y-25.359 to X32.757 Y8.619 with some odd extrusion in between.  This happens dozens of places in the resulting G-code (other examples @ line 2990,3454,3918)

````
G1 X-31.130 Y-25.359 E14.97306
G1 E12.97306 F2400.00000
G1 X32.757 Y8.619 F6000.000
G1 E14.97306 F2400.00000
G1 X32.710 Y8.986 E14.98396 F900.000
````

Slic3r config, STL and resulting Gcode here.  Zip File.
https://drive.google.com/open?id=0B0bthNAR9o1rYXBleEpqNTZjVzg&authuser=0

alexrj, 2015-05-03:
So, this is the hollow version of the stretchlet. Slic3r computes the medial axis for it, but it contains some minor tiny extrusions that confuse the vase algorithm. This has been fixed in 1.2.7-dev: if a layer doesn't contain a pure single loop, it's not spiralized.

The proper solution will be to improve the medial axis algorithm so that it returns a clean loop for a model like this or sorts its segments in a more continuous way. Spiral vase algorithm should also be more tolerant for travel moves during the loop.

The workaround is to use the solid version of the strechlet and set Fill Density to 0%, Top/Bottom Solid Layers to 0, Perimeters to 1.

alexrj, 2015-05-22:
Good news: this is now fully fixed.

closed 2015-05-22 by alexrj