slic3r/Slic3r #2773: Bottom solid layers not available [1.2.6]

cocoknight, 2015-04-05:
Happy Easter everybody,
I've been playing around with custom support directly generated in my CAD software. For these structures I want to apply special printing/slicing parameters. Works fine so far, but when I want to set the number of top and bottom infill layers Slic3r only gives me the option for the top layers.
[image]
Don't know if the option is hidden somewhere in the background or if there isn't a possibility so far to set that value. Would be happy if you could fix that issue.
Regards
Stefan

alexrj, 2015-05-02:
Are you sure? I see both:

[image]

cocoknight, 2015-05-03:
You are right with this. My bad! I've totally overseen that.

closed 2015-05-03 by cocoknight