slic3r/Slic3r #2774: Slicer crashes when the first part in the parts list is deleted [1.2.6]

cocoknight, 2015-04-05:
I just noticed that Slic3r is always crashing when I delete the first part in my parts list.
[image]
As long as I did not add any other part the delete button is grayed out. Adding a second part enables me to delete any of them. Deleting the second one in the list is fine. Deleting the first crashes Slic3r.

Cheers
Stefan

alexrj, 2015-05-02:
Can you please provide detailed steps in order to reproduce the issue?

cocoknight, 2015-05-03:
- Add a new part to Slic3r (hole.stl)
- RMB - Settings
- "Load part..." (add a second part. I my case "support.stl")
Now I have "hole.stl" and "support.stl" in the list. Now I want to replace the first part in the list by deleting it:
- Select the first part "hole.stl" - Delete part
--> Slic3r crashes on my machine

I hope you can reproduce that issue.

alexrj, 2015-05-03:
Okay, thank you. I'm able to reproduce this on Windows only. It looks like an ugly wxWidgets issue.

alexrj, 2015-05-04:
After a long debugging session I was able to find the wxWidgets bug causing this behavior, and I pushed a workaround. The issue is now fixed, and Slic3r does not crash anymore. Thank you @cocoknight for reporting this.

Docs for wxTreeCtrl state that the DeleteChildren method "will not generate any events unlike Delete() method". However, close inspection of the wxWidgets code reveals that DeleteChildren is implemented in MSW as a call to Delete(). So it will generate events, including EVT_TREE_SEL_CHANGED.

closed 2015-05-04 by alexrj