slic3r/Slic3r #2775: Decimal places in speed and layer offsetting

Lord2Vader, 2015-04-06:
Hi,

ive tested this from 1.1.7 to 1.2.6 and the results are the same:

This happens in slic3r:
G1 X77.797 Y109.381 E4.14597 F1200.000

And this in cura:
G1 F1200 X66.757 Y77.075 E0.01152

First of all the three decimalplaces after the F1200 are not very useful so they should not be written into the code.

And then putting the F on the end of the line, isnt this driving a ramp towards that speed from the last  speed? Why is this so and not in Cura?

The Problem is, with slicer and certain stls i get layeroffset and a certain height, with cura not!
Iam not sure but i think this speedingrampthingi is the reason, cause Cura even does harder moves/ changes and even on pushing the speed of my printer harder and harder i get no layeroffset with cura. So it should not be a mechanical or electrical problem.

I will post the STL and gCode later when i am home.

Lord2Vader, 2015-04-06:
Here the STL, gcode and pic here:
http://www.thingiverse.com/thing:758676

The STL is getting split in slic3r (or repetierhost). In Cura it works fine like I said.
Dont care about overhangs, happens in some stls with no overhangs.

The STLs are made with 123D Design

alexrj, 2015-05-02:
The order of parameters in a G-code line is not meaningful.

closed 2015-05-02 by alexrj