slic3r/Slic3r #2776: Space between perimeters

HcDevel, 2015-04-07:
I noticed a problem with the slic3r binary 1.2.6 experimental for Linux. After printing a part, it was not as stable as expected because of space between the perimeters.
I use a 0.4 mm nozzle, the wall should have been 1.6mm thick, so 4 perimeters should have been printed. Insted of that only 3, not connected perimeters.
Inspecting the problem using Pronterface I could see a hole which is placed in the wall in the top view. Trying the same again with Slic3r 1.1.7 stable everything worked as expected.
I uploaded the stl file, my configuration and the generated gcodes to http://blog.smartnoob.de/files/3d_print/slic3r_issue/. Both gcode files were generated using the same configuration

[image]
[image]

Lord2Vader, 2015-04-07:
Well first of all: this isnt a bug. 3D-gCode-Previews cant show your extrusionwidth, so if that changes, the space between the line changes ;)
And if you have an 1.6mm nozzle you get 3 perimeters ^^ In slic3r you MUST change your perimeter or default extrusionwidth to get 4 perimeters.

See gcode:

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.50mm
; infill extrusion width = 0.50mm
; solid infill extrusion width = 0.50mm
; top infill extrusion width = 0.50mm

As you can see ive changed the default width to 0.5 (for a 0.5 nozzle). Normally permeters extrusionwidth is wider than that.

SaintGimp, 2015-04-07:
Seems like this is the same problem as #2560, right?

HcDevel, 2015-04-08:
@Lord2Vader Okay, the perimeters weren't set correctly but there must be a bug since I got different results using the same configuration on different Slic3r versions. Either the last stable version creates to much perimeters or the latest experimental to few.
I think it would be a backstep when Slic3r would create thin walls which would make the hole thing instable just because the amount of perimeters wasn't set correctly.

@SaintGimp This might be the same problem but as mentioned, the problem appeared in the experimental and not in the stable. So it's reversed.

alexrj, 2015-05-03:
Don't trust 3D preview from external programs. They don't have the necessary information for drawing a proper rendering. Please only trust the 3D preview from Slic3r itself.

If you have airgaps between perimeters of printed objects, just increase the Extrusion Multiplier (and/or check your filament diameter and E steps) until you extrude enough plastic.

Different Slic3r versions might of course work in a different way. Otherwise no new versions would have been released ;)

HcDevel, 2015-05-03:
So it is wanted that there is space between the perimeters when there is set a limit of 3 perimeters with a 0.4 mm nozzle and the wall should be 1.6mm thick?

Since I also printed this I can say the 3D preview was correct

alexrj, 2015-05-03:
Well, you're relying on several wrong assumptions. I'll help you understand things better.

First, I'll repeat that no 3D preview tool is able to do a draw a proper rendering of extrusion widths because that information IS MISSING in the G-code file. Trust me, don't insist saying it's correct.

Second, you're counting perimeters the wrong way. Every shape having a hole (like, in your case, a rectangle having a rectangular hole thus forming a wall) has perimeters on both the external side and the internal side. So if you specify 4 perimeters you're asking Slic3r to fit 8 loops in there (4 on the external side and 4 on the internal side), but of course there's no room and Slic3r just does what it can.

Third, you rely on the wrong assumption that having a 0.4mm nozzle perimeter loops will be 0.4mm thick and will be spaced by 0.4mm. This is wrong for two reasons. As you can read in your own G-code file, Slic3r used a 0.4mm extrusion width only for the outermost loops (external) and used a thicker width for the internal ones (0.67mm). You can set custom widths in the Advanced section if you want. The other reason is that some overlap is generated, so perimeters will be spaced a little less than their nominal width.

If you look at the preview in Slic3r you'll understand what it's trying to do:

[image]

i.e. two external 0.4mm loops and one single fat fill trace in the middle.

If you have air gaps, as I said, you can increase the extrusion multiplier.  If you want 4 loops you can set the Perimeter Extrusion width to 0.5mm manually and you'll get this:

[image]

closed 2015-05-03 by alexrj

andreykyz, 2018-10-12:
@alexrj why did you close this issue? This is the real problem

https://shop.prusa3d.com/forum/others-archive--f82/gaps-in-infill-near-hole-perimeters--t3759.html
https://reprap.org/forum/read.php?263,755174
https://community.ultimaker.com/topic/9696-gap-on-between-perimeters-and-top-infill/

lordofhyphens, 2018-10-12:
@andreykyz I thought @alexrj explained himself pretty clearly in his comment. 

Do you have new evidence to rebut it?