slic3r/Slic3r #2777: Incomplete model results from what should be a clean slice

noahbnkatz, 2015-04-09:
Wanted to use 1.2.6 so I could use zero interface depth support material. That part looks GREAT, but the program crashes and fails to export a clean model. The top 20% of the model (the ford text) is either partially or fully excluded from the final gcode. Tried 3x. 

[image]

harriv, 2015-04-09:
It's impossible to solve this without the actual model file and Slic3r config.

Please read https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

Artesian3D, 2015-04-10:
I know that... how do you expect me to upload files that aren't pictures? The uploader only allows pictures.

zsolta, 2015-04-10:
You can try dropbox or similar and post the link here.

Artesian3D, 2015-04-10:
https://www.dropbox.com/s/5wi92sdm0y1uzpl/config%20-%20Artesian%204-9-15%20for%20Ford%20Logo.ini?dl=0

https://www.dropbox.com/s/2rmqe2fbnzy4h7i/Ford%20Front%20Emblem%20-%20for%20SW%20SUV-v1.stl?dl=0

There we go. 

*Emblem is a custom design I made for a client. Please don't steal. <3

alexrj, 2015-05-18:
Okay, the behavior you describe is a crash during G-code export. It may be caused by an out-of-memory error or another bug. Since I could not reproduce it I assume it's fixed in 1.2.7-dev. However I noticed the G-code export to be very slow because of *Avoid crossing perimeters*, so I pushed a change that speeds it up. I'll release 1.2.7 soon, let's test with that one.

alexrj, 2015-05-24:
@Artesian3D, does 1.2.7 fix this issue?

alexrj, 2015-06-20:
@Artesian3D, can you please check Slic3r 1.2.9 and confirm we fixed the problem you reported? Thank you!

alexrj, 2016-03-18:
@Artesian3D, no answer from you so I assume this is fixed. Feel free to reopen.

closed 2016-03-18 by alexrj