slic3r/Slic3r #2778: Feature Request : Purge / Prime Tower

SystemsGuy, 2015-04-10:
Number of new heads on the market or coming to market that support multiple materials extruded from a single orifice.  The use of these type of heads requires an active purge to clear one color for another.

Using the "tool change" gcode in slic3r today requires and active and designed "purge and clean" area where one can move the head, purge the material, and then clean the remnants off before resuming printing.  This is challenging at best.

Using the "tool change" gcode to manually build a prime / purge pillar is an option that require a tool change every layer.  This is not ideal as it requires a specific design component in the model to accommodate it.  It also requires specific code dependent on layer height.

An alternative to the above is to force a tool change every layer by using the secondary color as infill.  This is less than ideal as it forces unnecessary tool changes, and allows for bleed  through on translucent filaments. 

If there are other workarounds I would be interested to hear them.

jonaskuehling, 2015-04-21:
Thanks @SystemsGuy for bringing this up, didn't yet manage to file this request after our discussion in #1780

My strong support for this one. @alexrj I think this technique is currently the most effective, reliable and quite importantly most compatible solution for so many different printers, making for amazingly nice multi-extruder experiences, not only for the upcoming single-nozzle-multi-material heads, but also for classic multi-nozzle setups.

curiouspl2, 2015-05-29:
+1 on this one. it would be also cool to be able to load user stl file for the tower to allow making some use of the 'wiped' plastic

kyrreaa, 2015-06-14:
+1 and option to use mixing extruder instead of tool change...

gralco, 2015-06-26:
+1 and an ooze shield would be pretty sweet too! (maybe needs a separate request but still very relevant)

lordofhyphens, 2015-07-30:
in post-processing land, you can litter the gcode with placeholders for a script to read/understand and place the appropriate prime stuff there (like multiple lines if you are swapping multiple times). Once you get that point though might be worth trying to rewrite it in perl.

kyrreaa, 2016-08-23:
+1 for me too.
My new tripple input head was mostly for water soluble support but even then I need to purge to avoid mixing it in with the print.
A dissolving layer in a print would be bad...

Small volumes should be deposited as round or square tubes to give them better rigidity in case of a tall print.

An option to dump dump it in infill if possible would also be cool.

magu, 2016-08-29:
+1 on me as well. Got my E3d cyclops up and running, and getting it to prime the nozzle properly is a real pain, even with a few of the post processing-scripts available.

Jerimon1998, 2016-11-23:
+1, and probably +25 or so if I told the 3D printing class at my high school about this. I actually am unable to use the second on my extruder right now, because slic3r doesn't have any means to prepare the nozzle automatically, other than the quite useless tall skirt, which traps excessive amounts of heat and usually pops off mid-print anyways; and while Cura did generate the purge tower I needed, it refused to accept the settings I gave it (based on those in slic3r), and so the print failed catastrophically (print popped right off the bed after the first 5 layers).
My recommendation for implementing this is to add the feature in the "Ooze Prevention" section. Instead of forcing the tall skirt by default, add a drop-down menu with 3 (or more) options: Tall Skirt (the original), Hollow Towers (a hollow, square tower for each nozzle, placed off to the side), or Solid Tower (a single, solid tower for all nozzles).
I also have recommendations for the actual function of these towers:
When a tool change occurs, the old nozzle will go to the tower and wipe itself as it cools down. Then, the new nozzle comes over and extrudes a layer as it heat up.
For the solid tower, divvy it up into triangles of equal sizes for the number of tool changes necessary per layer. This may require a wider tower in  quad-extruder printers, but you could also create more towers and simply keep the max number of triangles per tower to 2.

Jerimon1998, 2016-11-23:
May I also mention that I have tried using the post-processing script [here](https://github.com/makertum/Multi-Extrusion-post-processing-scripts-for-Slic3r), but neither version actually generated a tower in the gcode, even after trying all the fixes that were recommended.