slic3r/Slic3r #2780: MORE Octoprint Integration!

flyswatter, 2015-04-12:
Closing issue #1826 is the coolest thing ever, Slic3r is very close to becoming the only application I need to 3d print!

If you just add a "Print" button, I won't even need to open the OctoPrint web interface!  I think this would be a small feature with a big enhancement.

Long term, it'd be cool to provide even more of a front-end, but it's realistically probably beyond the scope of what Slic3r is about..

lordofhyphens, 2015-04-12:
Disagree here. Slic3r should be about slicing models and generating gcode.

If you want a more integrated solution someone was working on a octoprint
plug in for slic3r.

flyswatter, 2015-04-12:
I didn't realize slic3r could take plug-ins!  That would be a cool project!  Looks like it's kept [over here](https://github.com/OctoPrint/OctoPrint-Slic3r/tree/master).

lordofhyphens, 2015-04-12:
Octoprint can, and you can do any sort of "postprocessing" script in slic3r.

alexrj, 2016-12-17:
I added a prompt to start printing at upload time.

closed 2016-12-17 by alexrj

flyswatter, 2016-12-17:
Cool, thanks!