slic3r/Slic3r #2781: Thin walls printed without infills making them weak

led42, 2015-04-13:
I've been using slic3r version 1.2.6 and noticed some models print with thin walls without any infill making them really flimsy.
Looked in previous issues and looks similar to what was happening in issue #281 I even used the rake stl file presented in that issue http://nypano.com/clients/UM/rake.stl (not my file) and looking at the preview on 1.2.6 3 of the walls are empty.

Models that also exhibit this problem: http://www.thingiverse.com/thing:604915 http://www.thingiverse.com/thing:423216

I'm using a 0.5mm nozzle, 0.1mm layer height.

led42, 2015-04-13:
The 3 walls on the left are empty, this is from the rake.stl file.
[image]

lordofhyphens, 2015-04-13:
Led42 please post entire configuration file.

led42, 2015-04-13:
Here it goes

# generated by Slic3r 1.2.6 on Mon Apr 13 21:39:52 2015

[filament:ABS]
bed_temperature = 105
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 1
extrusion_multiplier = 0.9
fan_always_on = 0
fan_below_layer_time = 999
filament_diameter = 1.8
first_layer_bed_temperature = 105
first_layer_temperature = 240
max_fan_speed = 100
min_fan_speed = 100
min_print_speed = 10
slowdown_below_layer_time = 5
temperature = 240

[filament:M-PLA-05-03-1mc]
bed_temperature = 57
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 1
extrusion_multiplier = 0.9
fan_always_on = 0
fan_below_layer_time = 999
filament_diameter = 1.75
first_layer_bed_temperature = 57
first_layer_temperature = 0
max_fan_speed = 100
min_fan_speed = 100
min_print_speed = 10
slowdown_below_layer_time = 5
temperature = 0

[filament:PLA 0.5]
bed_temperature = 57
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 1
extrusion_multiplier = 0.8
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1,74
first_layer_bed_temperature = 57
first_layer_temperature = 195
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
slowdown_below_layer_time = 30
temperature = 195

[filament:PLA-03]
bed_temperature = 57
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 4
extrusion_multiplier = 1
fan_always_on = 0
fan_below_layer_time = 30
filament_diameter = 1.75
first_layer_bed_temperature = 57
first_layer_temperature = 200
max_fan_speed = 100
min_fan_speed = 100
min_print_speed = 10
slowdown_below_layer_time = 25
[... 734 lines cut ...]
support_material = 0
support_material_angle = 0
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 200
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0
z_offset = 0

VanessaE, 2016-03-19:
Here's another situation where gap fill is not working:

Grab this Thing, http://www.thingiverse.com/thing:1271651 

For the sake of saving time, you might want to split it up and consider only the five gears on the right.

Notice the gears' teeth, and arguably around the center of the largest of the five.  No combination of settings causes Slic3r to fill them in.

Config bundle:  
[Prusa i3 - Slic3r 1.3.0 config bundle.ini.zip](https://github.com/alexrj/Slic3r/files/180996/Prusa.i3.-.Slic3r.1.3.0.config.bundle.ini.zip)

alexrj, 2016-03-19:
@led42, this is how rake.stl appears with your config as of 1.2.10-dev (but this was probably already fixed in the previous versions:

[image]

alexrj, 2016-03-26:
@VanessaE, I fixed that in both stable and master. However, those are very very thin gaps and I'm not sure how effective/useful that gap fill is physically speaking.

closed 2016-03-26 by alexrj

VanessaE, 2016-03-26:
It's angle-sensitive but yep, seems much better now for most of the gears in that test.