slic3r/Slic3r #2782: Idea for smoothing near-horizontal surfaces

whosawhatsis, 2015-04-13:
There's been talk recently of printing non-planar layers for (among other goals) getting smoother top surfaces. I had another idea for this. Rather than trying to print lines that don't follow the X/Y plane, you could lift the nozzle AND increased flow so that you could print on top of a flat layer and maintain the same extrusion width, but the layer thickness would increase in certain areas, up to twice the normal layer height, to create a more gradual transition leading up to the edge of the next layer's perimeter.

This would mean that movement outside of the X/Y plane would be limited to less than a single layer lift, which would remove most of the worry about hot end and gantry geometry that non-planar layers introduce. 

This would mean breaking up printing move segments to include Z axis movement, as spiral vase moves do. Ideally, the speed of these segments should be adjusted to try to keep the extrusion rate constant as the volume per millimeter of horizontal movement increases to (almost) double, so that extrusion speed delay isn't a factor.

whosawhatsis, 2015-04-13:
Here's an illustration of what I'm thinking:

[image]

lordofhyphens, 2017-03-19:
Actually, I think this is covered by #1386

closed 2017-03-19 by lordofhyphens