slic3r/Slic3r #2783: Fedora 21 slic3r compilation problem

didli, 2015-04-14:
Hi everyone !
All my compilation tentatives failed, from slic3r 1.2.3 to slic3r 1.2.6, always with the same error at the same point :
'''Could not create MYMETA files...
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from _build/prereqs ...
Failed to upconvert metadata to 1.1. Errors:
Missing mandatory field, 'version' (version) [Validation: 1.0]
 at /usr/local/bin/cpanm line 36.
The XS/C++ code failed to compile, aborting
'''
Does this mean something to you guys ? Any advice please ?
Thank you !

KyleCurrier, 2015-04-15:
Line 36 sounds familiar. I used to run into a few. Check to see if model height is divisible be layer height. That's thrown errors at me before. Also try increasing thread count. That also fixed some random errors for me. If you have a slow comp, that will speed it up a little bit. You can also go to task manager and set slic3r task priority to high by right clicking it, that will help speed things up too but it can cause lag if you're doing something else at the same time. Alternatively if you paste the error text into Google, it should come back with someone who has this issue, and usually a solution in a reply. I may be way off on what that error is from but I've had quite a few when I started out and hadn't had any since making those changes and checking to see the model was divisible.

didli, 2015-04-15:
uh ? ... > Check to see if model height is divisible be layer height.
Slic3r is not even installed, since I'm already struggling when trying to compile it from source (the error comes when the compilation reach XS), so I don't have access to its configuration. 
Thank you for your advice and yes, I've tried to google this error, but search results are quite unreliable : this error could happen for every software using perl (or something related ?). I think it's more a configuration problem, something i must perform before trying to compile slic3r, but i have no idea where I should look (thus, the question here).
Anyway, thanks for your kind answer :). Got to look into it a bit more i guess.

didli, 2015-04-15:
Solved ! There were quite some fixes to finally get it to work. I finally was able to compile (git) following what is below :
- In Fedora 21, I need to call cpanm first instead of sudo perl Build.PL : 
```sudo cpanm --installdeps . --verbose```
But then I end up with an error stating my CPAN::Meta version (1.4405) was not the one required (1.44xx-something). I then edited the file /usr/share/perl5/vendor_perl/CPAN/Meta.pm (line 60) and change the required CPAN (1.4414 bla bla) to mine (1.4405).
Restart my first command and ... success !
Finally, i compile for the --gui option using : 
```sudo perl Build.PL --gui```
Done ;)

closed 2015-04-15 by didli