slic3r/Slic3r #2785: Custom infill possible?

KyleCurrier, 2015-04-15:
Hey Alex. What are the chances of seeing custom infill in future builds? It would be nice to import a 2 color jpg or something like a dwg to define a path for infill, be it tilable or not. Or perhaps a grid module that allows us to select a path via highlighting grid boxes. I may donate for such an ability.

a4jp-com, 2015-04-15:
That's a fun idea ^^

kefir-, 2015-04-15:
I suggested something very similar in issue #942.

Lord2Vader, 2015-04-16:
Would be nice for art stuff yes!

Maybe with .svg files cause the use coordinates

jonaskuehling, 2015-04-22:
Nice idea in general, at least for artistic purpose.. But since custom infill patterns made from an image/svg won't be printing-path-optimized at all, I think it won't differ much from using the already existing (and amazingly powerful) "modifier part" functionality to use even 3D geometries rather than just 2D patterns to apply different infill settings (like 100% or 0%) to certain regions of the model.

To apply custom 2D infill patterns, just use the intended svg/dxf/2-color-image, convert it to paths, import them into your CAD and extrude into 3D. Now export to STL and use it as modifier part in Slic3r to manipulate the infill density in the intersecting regions to 100%, while the default infill for the modified part is set to 0%.. This should produce your desired custom artistic infill I guess.

Furthermore, you could apply different available infill patterns like honeycomb or concentric etc. to different regions of a part this way..

ML3D, 2015-04-30:
I also like the idea of importing an alpha image and/or path file directly into Slic3r to be used as an infill guide. This would be so sweet.

As for modifier meshes, it would be really helpful if they could be positioned directly in Slic3r. This way basic modifier shapes could quickly be applied to any location on the part.

lordofhyphens, 2016-05-29:
This is a duplicate of #942... so I'm going to close it. @jonaskuehling 's workaround is definitely neat though, but I suspect we would need a more robust UI for modifer meshes to not make it a complete and utter pain to use it for that. I think AMF would be a good format so if it auto-split on load for a modifier mesh.

closed 2016-05-29 by lordofhyphens