slic3r/Slic3r #2786: Multiple perimeter vase mode?

whosawhatsis, 2015-04-15:
I think vase mode would be a lot better, both in terms of water-tightness and possibly optical clarity if it was possible to make spiral vases with multiple perimeters. The idea is that you would break up the Z move across the total number of segments for all perimeters, instead of just one. This way, if you had two perimeters, it would alternate between the inner and outer one, with the two vertically offset by half a layer height. This would allow the plastic to pack hexagonally rather than rectangularly to better fill the space and give you a better chance of having optically clear walls when using transparent materials like PET.

Is this doable?

xoan, 2015-04-15:
It sounds interesting, but not related to Spiral Vase mode. Maybe something like "Alternating perimeter layers", because Slic3r don't need to do the toolpath by increasing Z segment by segment. Un fact, by doing more than one perimeters you allways have a break point.

Anyway, hexagonal packing sounds cool :)

whosawhatsis, 2015-04-15:
Yeah, alternating the height of perimeters on normal vertical surfaces should increase strength, but it makes things really tricky when the wall is at an angle. You'd probably have to do some dynamic thickness adjustment as I suggested in https://github.com/alexrj/Slic3r/issues/2782

xoan, 2015-04-15:
When I said "not related to Spiral Vase mode" I mean not related to the "spiral" part.

whosawhatsis, 2015-04-15:
Well, it is related, in this case, because doing this means that you're creating multiple walls from a single "spiral" (actually a helix). Z movement is continuous, just the offset from the outer surface changes.

xoan, 2015-04-15:
OK, so I don't understand :D 

What's the difference between doing this in spiral mode (actually helix is a better, better name) or doing this by puting half-layer height inside perimeter, then raise Z half-layer height and puting outside perimeter? I think using this way is more efficient.

whosawhatsis, 2015-04-15:
The point of spiral vase is that you don't need to stop to do a Z transition. Of course, scheme would involve a horizontal fast travel move (ideally crossing at 90 degrees at half extrusion) to cross between the inner and outer perimeter, which might result in a bit of a seam, but should still give better watertightness than a single wall, as well as a more solid wall that should offer better optical characteristics.