slic3r/Slic3r #2787: [1.2.7-dev] [Windows] Errors reading preset files

nicoruy, 2015-04-16:
After saving a preset slic3r shows a set of error messages and crashes.
This happens both on basic and expert mode.
On Basic mode the messages are presented when the application starts.
On Expert mode it happens when the save button is pressed on the settings tabs.
You can see the errors on the attached screenshots.
The first message is presented twice.

The problem might be related to the issue #2645 (I have special characters on my path)
[image]
[image]
[image]

alexrj, 2015-05-06:
@nicoruy, should should be fixed and I can't reproduce it (I tested on a Windows 7 machine having username with a "à" letter in it)... Since you didn't specify the commit you were running at the time of the report, could you please `git pull` now and `perl Build.PL` (making sure the build succeeds) and let me know if this still happens? Thank you!

clefranc, 2015-05-21:
@alexrj  I reproduced this problem using the celcius character in the preset: **Test Temp 230°C**.
Exact same dialog messages, except in French. Hope this help!

alexrj, 2015-05-24:
@clefranc, what Slic3r version?

clefranc, 2015-05-24:
@alexrj 1.2.6

alexrj, 2015-05-24:
@nicoruy and @clefranc: can you check whether 1.2.7 (released today) fixes this? Thank you!

clefranc, 2015-05-25:
@alexrj Same error with 1.2.7.

When using Repetier-Host, I can use the **Test Temp 230°C** to generates g-code without error.

alexrj, 2015-06-09:
@nicoruy, @clefranc, could you please try with 1.2.8? This should be fixed now. Thank you

alexrj, 2015-06-20:
@nicoruy, @clefranc, could you please try with 1.2.9 and confirm this was fixed? Thank you!

clefranc, 2015-06-21:
@alexrj Sorry for the delay, back from vacation...

Yes, I confirm it's fixed with 1.2.9. Thank you!

alexrj, 2015-06-21:
Thank you @clefranc!

closed 2015-06-21 by alexrj