slic3r/Slic3r #2789: Configuration

koclukasz, 2015-04-17:
I have a problem: Every time i start slic3r it looks like i do it for the first time, configuration wizard shows and any configuration i set before is lost. Moreover when i checked expert mode in setup and restarted slic3r it started in simple mode. Any ideas how to fix this?

gege2b, 2015-04-19:
looks like  some write permissions troubles on the folder where slic3r saves his config

more info on the OS ?

koclukasz, 2015-04-19:
it is normal that repetier folder is "read only" ?
windows 8.1 dell inspiron5537

gege2b, 2015-04-20:
I don't know, I don't use repetier at all
If it's installed on the "program files" directory, then maybe its normal

maybe you can specify a path in repetier for the slic3r config ?

alexrj, 2015-05-02:
I don't really know how Repetier handles Slic3r. Try to contact Repetier's author or issue tracker. Try to give write permissions to the Repetier and Slic3r directories.

closed 2015-05-02 by alexrj