slic3r/Slic3r #2790: Slicer does nothing

pOmelchenko, 2015-04-17:
Hi! Yesterday noticed a problem with a long work slicer.

Dispatched slicer model and for a long time export gcode do nothing. Wherein the disk file is not increased.

Here is a model that tried to cut

http://www.thingiverse.com/thing:762168/#files

However, other models are cut properly. This model is opened and exported to kisslayser normal. Without mistakes.

pOmelchenko, 2015-04-17:
http://pastebin.com/banveS6f

backup config

pOmelchenko, 2015-04-17:
Found the cause of the problem, the thing functions "avoid crossing perimeters". I turned off this feature, the file is cut in seconds

lordofhyphens, 2015-04-17:
Would also have been very helpful to know the version you were using. There
are some known issues w/r/t "Avoid Crossing Perimeters".

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

pOmelchenko, 2015-04-18:
I tested and stable 1.1.7 and 1.2.6 on a pilot.

As it turned out it it is certainly doing something, but it does so very slowly.

After I cut model (link above) without this feature, the size of about 14MB gcode.

And when you use this function the file size up to 1.5Mbit grew 30 minutes.

pOmelchenko, 2015-04-20:
Today, the compiled version 1.2.7, the problem with the Save function. STL file slice for a long time, while vyrostaet total consumption of system resources.

VanessaE, 2016-06-29:
This is still a problem with current code.  To gauge the speed, give this STL a try:

http://www.thingiverse.com/thing:1271651/#files

The initial trangulated mesh and perimeters steps take a bit, but when it goes off into the final exporting-to-gcode step, it just .... stops.  It does eventually finish exporting, and the result looks okay in Pronterface (loaded it there just to see the resulting travel paths).

lordofhyphens, 2016-06-29:
@VanessaE did you check with the PR for streamlined Gcode export by
@bubnikv or just 1.2.9 or just current master?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

VanessaE, 2016-06-29:
Just current master; I didn't remember that PR.

lordofhyphens, 2016-06-29:
Okay, just wanted to be sure.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

VanessaE, 2016-06-29:
That PR does offer some improvement (I ran these tests multiple times; for each test, I cleared the plate, re-added the STL, and exported, to force a complete re-slice).  Using the calipers STL previously linked:

With a clean clone of upstream, and "avoid crossing perimeters" not checked, it takes around 5½ minutes for the program to do a full slice-and-export:  a bit under 4 minutes for the stuff leading up to the final export-to-gcode step, and 1½ minutes for the export.

With the g-code PR applied (on top of a fresh clone) and "avoid" still turned off, it took 3 to 5 minutes to run through the whole process: just shy of 2 minutes for the lead-up stuff, and 1 to 3 minutes to finish the export step.

With clean upstream, if I enable the "avoid" checkbox, it takes a bit under 4 minutes for the lead-up stuff, and 13 to 29 minutes for the export step (in one case, I canceled and started over).

With the PR in place and "avoid" enabled, under 2 minutes for the lead-up, and 13 to 13½ minutes for the export step.

lordofhyphens, 2016-06-29:
@VanessaE Thanks, seems like we can link this issue to #3355 then.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-06-30:
The "avoid crossing perimeters" is indeed a huge bottleneck in the g-code generation. If Prusa3D used a bowden extruder, this would probably be the 1st candidate for me to rework.
Vojtech