slic3r/Slic3r #2792: Extrusion problem with volumetric firmware

Alex9779, 2015-04-17:
Hello,
I recently started using slic3r, after giving Cura its chance and configuring Simply3D to print good results on my printer, to have more options on complex parts.
Though I am having trouble with the initial settings to get a simple print correctly.
After doing some test prints I realised that my extruder constantly maxes out and the firmware retracts a bit so I compared the gcode of the same part in those three softwares and I saw that the extraction of slic3r is way too much than the others.
Here are my settings:
´´´
; avoid_crossing_perimeters = 1
; bed_shape = 0x0,230x0,230x215,0x215
; bed_temperature = 60
; before_layer_gcode = 
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_width = 0
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 1
; duplicate_distance = 6
; end_gcode = ;End G-code\nG91 ; relative coordinates\nG1 Z0.2 ; retract lift\nG1 E-20 F3000 ; reverse priming\nG90 ; absolute coordinates\nG28 ; home all axes\nM104 S0 ; turn off extruder heaters\nM140 S0 ; turn off bed heater\nM84 ; disable motors
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 1
; fan_below_layer_time = 60
; filament_diameter = 2.85
; first_layer_acceleration = 0
; first_layer_bed_temperature = 60
; first_layer_extrusion_width = 125%
; first_layer_speed = 60%
; first_layer_temperature = 210
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; infill_acceleration = 0
; infill_first = 0
; layer_gcode = 
; max_fan_speed = 100
; min_fan_speed = 80
; min_print_speed = 10
; min_skirt_length = 0
; notes = 
; nozzle_diameter = 0.4
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 0
; resolution = 0
; retract_before_travel = 2
; retract_layer_change = 0
; retract_length = 2
; retract_length_toolchange = 10
; retract_lift = 0.05
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 40
; skirt_distance = 6
; skirt_height = 1
; skirts = 1
; slowdown_below_layer_time = 15
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = G28 ; home all axes\nG1 F12000 X5 Y10 ; Cura HEATUP_POSITION_COMMAND\nM190 S[first_layer_bed_temperature] ; wait for bed temperature to be reached\nM109 S[first_layer_temperature] ; wait for temperature to be reached\nG1 F2400 Z20 ; bring plate to Cura PRIMING_HEIGTH with Cura HOMING_FEEDRATE\nG92 E0 ; zero the extruded length\nG1 F500 E20 ; undo end of print/abort retraction\nG92 E0 ; zero the extruded length\nG1 F50 E5 ; additional priming\nG1 F2400 X20 Y20 Z0.3 ; wipe start position\nG92 E0 ; zero the extruded length\nG1 F225 X35 E2 ; move 15mm and extrude\nG92 E0 ; zero the extruded length\nG1 F2100 X50 E1 ; move 15mm\nG92 E0 ; zero the extruded length
; temperature = 210
; threads = 2
; toolchange_gcode = 
; travel_speed = 130
; use_firmware_retraction = 1
; use_relative_e_distances = 0
; use_volumetric_e = 1
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.2
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 0.1
; raft_layers = 0
; seam_position = random
; support_material = 0
; support_material_angle = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_layers = 3
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_pattern = pillars
; support_material_spacing = 2.5
; support_material_speed = 60
; support_material_threshold = 0
; xy_size_compensation = 0
; bottom_solid_layers = 6
; bridge_flow_ratio = 2
; bridge_speed = 25
; external_fill_pattern = rectilinear
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 50%
; fill_pattern = rectilinear
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 60%
; infill_speed = 75
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 50
; perimeters = 2
; small_perimeter_speed = 30%
; solid_infill_below_area = 10
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 40%
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 30%
; top_solid_layers = 6
´´´

Maybe there is some calculation error when calculation E rate for volumetric extrusion? Because I have to set my extrusion multiplier to ~0.1 to get reasonable results and that is way off the normal values 0.9-1.1... it is a tenth!

Regards,
Alexander

alexrj, 2015-05-02:
Is your firmware configured to use volumetric extrusion by default? I don't see any `M200` command in your custom G-code.

Alex9779, 2015-05-03:
As far as I know it is.
In Simplify3D I also have no M200 lines, so I assume the firmware is set to volumetric by default. It's an Ultimaker 2, latest firmware...
I did three slices of the same model with slic3r, Cura and Simplify3D and compared the total amount extruded per layer. Cura does not reset the extrusion amount over the whole model, S3D does reset the amount on each layer.
As mentioned when I use an extrusion multiplier of 0.1 instead of 1 the values slic3r extrudes per layer are around the same Cura and S3D extrude.

whosawhatsis, 2015-05-03:
Ultigcode mode, which is where a UM2 uses volumetric extrusion and firmware retraction from settings stored in EEPROM, needs to be triggered by special comments in the start code. You need to have a line that says ";FLAVOR:UltiGCode", and I believe there are a few other lines that would otherwise be considered comments that a UM2 interprets in this mode.

This also means that you need to send the gcode using something that won't strip comments.

alexrj, 2015-05-03:
@Alex9779, comparisons are not helping much. Let's check whether the G-code produced by Slic3r is correct or not, regardless of other software. Can you paste the first, say, 50 lines of a G-code file? We can easily calculate whether the volumetric values are reasonable or not.

Alex9779, 2015-05-03:
@whosawhatsis Sorry to say but I am not a full noob in this. I know that Ultimkaer CAN use UltiGcode but does not have to. I am successfully printing with normal Gcode generated by Simplify3D, my own start and end sequence even if I start from SD card and also after a Cura sliced model with UltiGcode was printed, so my start and end sequence seem to be pretty decent.

@alexrj  I will create the test again, and post what you requested...

Alex9779, 2015-05-03:
```
; generated by Slic3r 1.2.6 on 2015-05-04 at 00:20:31

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm

M107
G28 ; home all axes
G1 F12000 X5 Y10 ; Cura HEATUP_POSITION_COMMAND
M190 S60 ; wait for bed temperature to be reached
M109 S210 ; wait for temperature to be reached
G1 F2400 Z20 ; bring plate to Cura PRIMING_HEIGTH with Cura HOMING_FEEDRATE
G92 E0 ; zero the extruded length
G1 F500 E20 ; undo end of print/abort retraction
G92 E0 ; zero the extruded length
G1 F50 E5 ; additional priming
G1 F2400 X20 Y20 Z0.3 ; wipe start position
G92 E0 ; zero the extruded length
G1 F225 X35 E2 ; move 15mm and extrude
G92 E0 ; zero the extruded length
G1 F2100 X50 E1 ; move 15mm
G92 E0 ; zero the extruded length
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.200 F7800.000
G10 ; retract
G92 E0
G1 Z0.250 F7800.000
G1 X102.244 Y97.229 F7800.000
G1 Z0.200 F7800.000
G11 ; unretract
G92 E0
G1 X102.929 Y96.544 E0.03432 F2160.000
G1 X103.091 Y96.406 E0.04186
G1 X103.875 Y95.836 E0.07619
G1 X104.057 Y95.725 E0.08373
G1 X104.920 Y95.285 E0.11806
G1 X105.117 Y95.203 E0.12560
G1 X106.039 Y94.904 E0.15993
G1 X106.246 Y94.854 E0.16747
G1 X107.203 Y94.702 E0.20179
G1 X107.900 Y94.666 E0.22651
G1 X122.100 Y94.666 E0.72941
G1 X122.585 Y94.685 E0.74659
G1 X122.797 Y94.702 E0.75413 F2160.000
G1 X123.754 Y94.854 E0.78846
G1 X123.961 Y94.903 E0.79599
G1 X124.883 Y95.203 E0.83032
G1 X125.080 Y95.285 E0.83786
G1 X125.943 Y95.725 E0.87219
G1 X126.125 Y95.836 E0.87973
G1 X126.909 Y96.405 E0.91406
G1 X127.071 Y96.544 E0.92160
G1 X127.756 Y97.229 E0.95592
G1 X127.895 Y97.391 E0.96346
G1 X128.528 Y98.279 E1.00208
G1 X129.016 Y99.220 E1.03964
G1 X129.097 Y99.417 E1.04718
G1 X129.397 Y100.339 E1.08151
G1 X129.446 Y100.546 E1.08905
G1 X129.598 Y101.503 E1.12337
G1 X129.634 Y102.200 E1.14808
G1 X129.634 Y112.800 E1.52349
G1 X129.615 Y113.285 E1.54067
G1 X129.598 Y113.497 E1.54821
[... 168 lines cut ...]
; fill_angle = 45
; fill_density = 50%
; fill_pattern = rectilinear
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 60%
; infill_speed = 75
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 50
; perimeters = 2
; small_perimeter_speed = 30%
; solid_infill_below_area = 10
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 40%
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 30%
; top_solid_layers = 6
```

Alex9779, 2015-05-03:
OK seems I was totally wrong... Sorry for bothering...

closed 2015-05-03 by Alex9779

alexrj, 2015-05-03:
@Alex9779 what was the error?

Alex9779, 2015-05-03:
I have to verify first but @whosawhatsis comment made me think again and I think my assumption was wrong: when I use normal gcode and not UltiGcode nothing from the firmware settings is used. S3D does not have a volumetric mode. I do not find any hint searching the web on that but it seems it only uses the normal feed mode. When I slice the model again with slic3r and volumetric setting off the extrusion values are reasonable again.
So you are right, when using volumetric I have to have M200 setting in my start code. Again, I have to verify first, have a print running atm but I am pretty sure that was my fault.

Think that is the problem when using such a "user friendly" printer, if you wanna use something different than what is meant to be used (Cura) you have to read more than if you would do it with the normal hard learning curve...