slic3r/Slic3r #2793: Slicer experimental crashes with following code

MaiMariarti, 2015-04-20:
Problem Event Name:	APPCRASH
  Application Name:	slic3r.exe
  Application Version:	1.2.6.36
  Application Timestamp:	503fe867
  Fault Module Name:	perl514.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	4f863eff
  Exception Code:	c0000005
  Exception Offset:	000000000003e61d
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	1033
  Additional Information 1:	7c43
  Additional Information 2:	7c439af64f5da66cd7968085b04ba711
  Additional Information 3:	c935
  Additional Information 4:	c935559384792a549cd91a82e1b3c8d1

Running windows ultimate x64

lordofhyphens, 2015-04-20:
So what were you doing? Please read the contribution guidelines and update
this issue, as it is impossible to know what is going on from the limited
information given.

MaiMariarti, 2015-04-20:
Sorry. my first time every reporting a bug.
Don't know enough about computers to follow up. The info I gave you is from the error when I try to slice ANY object in 3D.
Consider this thread closed. Maybe some one more knowledgeable can open it again. Love the 3D preview feature. Only if it didn't crash :(

closed 2015-04-20 by MaiMariarti

lordofhyphens, 2015-04-20:
What kind of video card are you using? What is the config in slic3r?

a4jp-com, 2015-04-20:
Can you try the newer version of Slic3r @MaiMariarti ? 

version 1.2.7-dev is very nice and hasn't crashed on my PC vet ^^ The older versions of Slic3r used to crashed a lot though.

MaiMariarti, 2015-04-20:
I have version : slic3r-mswin-x64-1-2-6-experimental
is there a newer version than this?
Cheers,
Mai

a4jp-com, 2015-04-20:
1.2.7
Did you download for the site or compile the software?

MaiMariarti, 2015-04-20:
Download. I have no clue how to compile it from source.

lordofhyphens, 2015-04-20:
I didn't think 1.2.7 had a precompiled windows version out yet.

a4jp-com, 2015-04-20:
The version I have says 1.2.7-dev.
It's the developers version but it has all the updates in it.
I can try zipping the files and hopefully you can run it.
If it is difficult I can slowly explain how to compile from the source code over the next few days if you like.

MaiMariarti, 2015-04-20:
If I can get the zip file, I can try that first?

a4jp-com, 2015-04-20:
Try downloading this from my server:

http://a4jp.com/3d-files/slic3r/slic3r-1-2-7-dev---a4jp-com.zip

I think it will work because I compiled the files and also made a shortcut in the folder. I just have this folder on the C drive.

MaiMariarti, 2015-04-20:
there are no .exe files for windows!
Do I have to compile this thing?

a4jp-com, 2015-04-20:
Maybe you have to load Strawberry Perl to get it to work. The shortcut in the folder points to the exe. Perl... Sorry I forgot about that. I was always just clicking on the link I made.

MaiMariarti, 2015-04-20:
no problem man. Thanks for trying to help. I appreciate it.
cheers

a4jp-com, 2015-04-20:
C:\Strawberry\perl\bin\wperl.exe C:\Slic3r-4\slic3r.pl

That is inside my loader.

Perl 32-
[image]
bit works best for me.

MaiMariarti, 2015-04-20:
It works perfectly :)
I love the gcode it generates. Very smart.
I wish we could save the plate arrangement instead of being able to only
generate STL and save the gcode.
Oh, their is hope.
Thanks again.