slic3r/Slic3r #2794: [1.2.6] Complete Individual Objects moves to new location, homes z, then jumps back to previous object

Shiral446, 2015-04-21:
Version 1.2.6
OS: Windows 8.1

**Problem:**
I was printing 6 calibration objects, and when one object would complete, it would move the X/Y to new location, home Z-axis, then jump back to the previous object for a command (knocking the build plate) before returning to the new object.

**Steps to Reproduce:**
Import two objects, and set to Complete Individual Objects. (I was using the SFACT thin wall calibration)

**Relevant G-code:**
G92 E0
G1 X16.155 Y51.364 F6000.000
G1 X18.155 **Y19.381** F6000.000     **<--- Moves to new location (note Y axis)**
M140 S90 ; wait for bed temperature to be reached
M104 S235 ; set temperature
G1 Z0.200 F6000.000                      **<---- Homes Z Axis**
G1 X21.980 **Y131.692** F6000.000              **<---- Jumps back to old location**
G1 X28.766 Y50.364 F6000.000
G1 X37.855 Y19.681 F6000.000
G1 E1.00000 F1800.00000
G1 X37.855 Y39.384 E1.26640 F1800.000

c-born, 2015-04-21:
Looks like issue #2691, which was fixed in 1.2.7-dev

Shiral446, 2015-04-21:
You would be correct, I mistakenly still had the is:open filter when searching it is was already reported. Thank you!
Closed!

closed 2015-04-21 by Shiral446