slic3r/Slic3r #2796: [feature request] exclude external perimeters from cooling related slowdown

curiouspl2, 2015-04-21:
i've noticed that when cooling is enabled, external perimeters - printed at various speeds - differ slightly on apperance. 
this produces 'banding' effect, though it's mainly visual, as actual size differences are sub-milimeter (hardly measurable by any of instruments i own).

If object is printed without cooling, everything goes visually perfect. 

I guess simplest method to overcome this problem would be to add option to simply exclude external perimeters from cooling (and make them print at exactly same speed on each layer) . 

sorry i do not post files/photos on this one as i think everyone encountered this already, and i don't have good macro/examples handy today - will probably care to update bit later this month. 
greetings

lordofhyphens, 2017-03-02:
From @curiouspl2 #3273 
>autocool is very cool feature, but it causes external perimeter speed to vary.
>this causes uneven finish of prints.
>
>it would be nice if external perimeters could be excluded from autocool, and printed at their nominal speed each time.
>perhaps another 'last resort' or 2nd pass could be added to slow down external perimeters when lowest autocool speed was reached on other features, so priority could be set this way.

From @curiouspl2 #3079:
>autocool feature is just cool, but as at different speeds of printing slight variations of pressure
and way plastic is laid down cause external perimeters to appear slightly different.
>This gives impression of 'uncalibrated' or 'random z wobble' print visually.
>One of solutions to the problem would be option to exclude external perimeters from autocool except completely nessesary (last resort) . This would require autocool to be performed in two phases,
>first excluding external perimeters completely, and then - if layer print time is still below threshold - performing autocool only on external perimeters remaining.

alexrj, 2017-03-03:
Done!

closed 2017-03-03 by alexrj

Tinchus2009, 2017-03-03:
I have just buuld the latest dev version and Im checking this new feature. This feature is automatic or there is an option? because I cant find the option in the GUI

lordofhyphens, 2017-03-04:
Should be automatic.

hyperair, 2017-07-15:
I just ran into an issue caused by this fix: my internal perimeters run at 5mm/s (minimum speed), and external perimeters run at 70mm/s (original speed), so I'm getting severe underextrusion at the beginning of the external perimeters.

curiouspl2, 2017-10-24:
Try enabling Linear Advance in Marlin. this should compensate for pressure differences in such edgy cases. also since your model already needs to be printed much slower , just reduce maximum speed. 

Also with Linear Advance enabled you can change acceleration settings, making speed transition much less abrupt.

hyperair, 2017-10-25:
I'm using Smoothieware which doesn't have pressure advance implemented yet (I mean there's an unmerged PR, but that's it). Printing slower always works, of course, but the cooling slowdown was a nice and convenient feature that allowed you to keep your speeds high for the larger layers in your print. It just becomes utterly useless if your external perimeter speed varies too much from the cooling slowdown's minimum speed.

I think this change may have been reverted though -- I'm not noticing this behaviour in my build (1.3.0~dev.git20171002.r1+a28f431d-1).

curiouspl2, 2017-10-28:
personally i opt for leaving it as an option (just a checkbox like 'exclude external perimeters from autocool' in the cooling section), and it would be nice if user could also set up exclude threshold delta to prevent artifacts. 

It seems hard to satisfy all the possibilities so i wonder if it wouldn't be nice to think of some scripting logic for options , allowing users to customize slicer options more to their printers.

DerSchultze, 2018-01-06:
Yep.. having this as an option would be really nice, I'm seeing the same issues as hyperair, with severe under extrusion on perimeters, when autocool kicks in.