slic3r/Slic3r #2797: [Feature Request] - insert custom code if layer time/volume below a set value

shadow651, 2015-04-21:
Ok, So we all know there's those prints where they get super small at the top. and no amount of cooling helps because the hot end is sitting in the same spot for every layer.

Currently, you can get these to print fine if you print 4 or 5 at once, because the printer will go away to work on the other ones, letting it cool.
Or you can have the printer run a custom go away and wait command between every layer (unnecessary for the lower layers)
Or, you can modify the g code by hand to add something between the layers in question (tedious, and prone to mistakes).

But wouldn't it be nice if we could insert custom code, say: go up 0.1mm, go away, wait 15 seconds, return, go down 0.1mm.
But only insert it that layers area, or time is below a value we can set?

Lord2Vader, 2015-04-22:
Good idea had problems with that even with big objects with small hulls

curiouspl2, 2015-04-23:
Calling some custom gcode, like start/stop gcode - would be very nice solution, or just emitting lines for post-processor like you do suggest. 

I think sound removed this skeinforge plugin (it was called 'orbit') because of mixed results for many people. 
main problem was that pressure in nozzle did drop and then restarting extruder caused underextrusion (recent pressure advance solves this) and another problem was the ooze. 
Depending on printer design one could just move to wipe position slowly, and then back from wipe quickly , doing pressure advance to restore lost pressure. 

personally when printing such objects i just slow down max possible, and then manually reduce nozzle temperature. even then even short 'parking' of nozzle (like cura does) helps if object is very small, gives precious few seconds preventing total meltdown :)

lordofhyphens, 2017-03-19:
Should be doable by modifying the cooling logic (for time) to add your gcode.

Volume would be a bit trickier (assuming you're talking about model volume), although you could probably handle *extruded* volume without too much effort.