slic3r/Slic3r #2798: support begins in the air.

lonelymyp, 2015-04-22:
when setting the angle of 17 degrees overhanging
[image]

stl: http://www.thingiverse.com/download:238391
gcode: https://drive.google.com/file/d/0B2sGop41unyObEpSSWhlaFh4Tkk/view?usp=sharing

lordofhyphens, 2015-04-22:
Could you export and post the entire config and indicate which version of
slic3r that you are using?

Zuborg, 2015-04-23:
Its very close to the my issue [#2618](https://github.com/alexrj/Slic3r/issues/2618)

lonelymyp, 2015-04-23:
Yes, I have the same thing happened on other models.
I'll try to disable detection of bridges.

lonelymyp, 2015-04-23:
slic3r 1.2.6
Config: https://drive.google.com/file/d/0B2sGop41unyOV2d1SVFoZFdsNDg/view?usp=sharing
Support of bridges does not affect

Another strange support:
[image]

MartinMajewski, 2015-06-03:
Same thing here. Unfortunately a quick search didn't show me this issue before, so I opened a new ticket.