slic3r/Slic3r #2800: Thin wall generation inconsistent across Z-levels - wedge example

BurninSun, 2015-04-25:
I've been having issues with thin wall generation and I've boiled it down to a relatively simple example - a wedge. The model should generate consistent gcode across all Z-levels but the thin walls at the tip of the wedge are seemingly random. With thin walls disabled, the model slices exactly how I expect - down to a minimum of 2 perimeter lines then ignoring the thin part of the tip.

Picture with 1000 words (Slic3r 1.1.7):
[image]

Slic3r settings: http://pastebin.com/v3ubs9Fy
STL file: http://pastebin.com/Q7DqVUx0
Generated gcode: http://pastebin.com/672rUpGp

Version 1.2.6 produces notably different but still "random" results:
[image]

alexrj, 2015-05-13:
Thank you for the report! This is now fixed, and I think this commit also fixes some other issues that were reported here in GitHub.

[image]

closed 2015-05-13 by alexrj

BurninSun, 2015-05-14:
Thank you very much. While this did fix one of the major issues I had, it also created a few more (or at least somewhere between 1.2.6 and now created a few more issues). More reports to follow :)