slic3r/Slic3r #2801: Git Master segfault using quickslice from gui

didli, 2015-04-27:
This seems to be duplicate of this one :
https://github.com/alexrj/Slic3r/issues/2468
Trying to use quickslice feature from gui ends up in segfault (Fedora 21 x64). There's no issue for the same stl files using the command line. Perl is 5.18 (i think) in F21.
Here's a backtrace : http://didli.free.fr/Images/Forums/slic3r_backtrace.txt
STL file : http://didli.free.fr/Images/Forums/ramps_fan_40mm.stl

alexrj, 2015-05-04:
Are you able to run Slic3r under valgrind?

didli, 2015-05-05:
Hi alexrj ! 
Unfortunately, i didn't even know valgrind existed before you asked about it. I've tried but I'm not sure it worked, using this command (without --suppressions=opengl.supp, because otherwise valgrind didn't seem to work at all)
valgrind --gen-suppressions=all --leak-check=full --num-callers=40 --error-limit=no -v perl /media/Databank/Apps/3dprint/slic3r-git/slic3r.pl --gui
Surprisingly, I was able to slice something using the gui under valgrind (but it's slow as hell).
Here's the log : http://didli.free.fr/Divers/debug_slic3r.txt (6.3 Mo txt file !)
If you want to suggest a command to run for valgrind, please ask :)
Note that running slic3r from gui again still cause the core dumped. 
On a more positive note, I really like to use slic3r from the command line (I only use the gui to setup settings). I've been able to make a custom thunar action for slic3r, making slice in a matter of seconds. Slic3r is really a good app, thank you for it !

alexrj, 2015-05-05:
Thank you for running under valgrind. This gives some hint but the bug is in Wx or in perl threading. I'll try to investigate.

This is also related to #1636.

alexrj, 2015-05-05:
I pushed a commit. Can you check whether it fixes the issue for you?  Sadly, I can't reproduce the crash...

alexrj, 2015-05-06:
@wolfmanjm reported this doesn't fix the issue for him.

Can you try with the 1.2.6 binary package downloadable at http://slic3r.org/download ? I use perl 5.14 for those packages, and I wonder this might be a 5.18 issue. This would explain why I'm not able to reproduce it...

alexrj, 2015-05-06:
Note to self, this is the relevant excerpt from @didli's valgrind output:

```
==7897== Invalid free() / delete / delete[] / realloc()
==7897==    at 0x4A07CE9: free (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==7897==    by 0x3A1E8A58AC: ??? (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E8A60CD: Perl_mg_free (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E8CDABC: Perl_sv_clear (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E8CE15C: Perl_sv_free2 (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E8C52D6: ??? (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E84F1F1: perl_destruct (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0xC800439: S_ithread_clear (threads.xs:238)
==7897==    by 0xC801572: XS_threads_detach (threads.xs:1394)
==7897==    by 0x3A1E8C3E84: Perl_pp_entersub (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E8BC215: Perl_runops_standard (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0x3A1E84BF84: Perl_call_sv (in /usr/lib64/libperl.so.5.18.4)
==7897==    by 0xF88F4CC: wxPliEventCallback::Handler(wxEvent&) (e_cback.cpp:93)
==7897==    by 0x33758F58EB: wxEvtHandler::ProcessEventIfMatches(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (in /usr/lib64/libwx_baseu-2.8.so.0.8.0)
==7897==    by 0x33758F5C3E: wxEvtHandler::SearchDynamicEventTable(wxEvent&) (in /usr/lib64/libwx_baseu-2.8.so.0.8.0)
==7897==    by 0x33758F5D11: wxEvtHandler::ProcessEvent(wxEvent&) (in /usr/lib64/libwx_baseu-2.8.so.0.8.0)
==7897==    by 0x32D6E79234: ??? (in /usr/lib64/libwx_gtk2u_core-2.8.so.0.8.0)
==7897==    by 0x336B60FC54: g_closure_invoke (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B6219E1: ??? (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B62A120: g_signal_emit_valist (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B62A34E: g_signal_emit (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x32D6275D45: gtk_widget_activate (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D615CA3C: gtk_menu_shell_activate_item (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D615CE3C: ??? (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D614782C: ??? (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x336B60FC54: g_closure_invoke (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B6218D9: ??? (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B629CF7: g_signal_emit_valist (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x336B62A34E: g_signal_emit (in /usr/lib64/libgobject-2.0.so.0.4200.2)
==7897==    by 0x32D62771BB: ??? (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D6145AD3: gtk_propagate_event (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D6145F7A: gtk_main_do_event (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D5C6064B: ??? (in /usr/lib64/libgdk-x11-2.0.so.0.2400.27)
==7897==    by 0x3369A497FA: g_main_context_dispatch (in /usr/lib64/libglib-2.0.so.0.4200.2)
==7897==    by 0x3369A49B97: ??? (in /usr/lib64/libglib-2.0.so.0.4200.2)
==7897==    by 0x3369A49EC1: g_main_loop_run (in /usr/lib64/libglib-2.0.so.0.4200.2)
==7897==    by 0x32D6144EC6: gtk_main (in /usr/lib64/libgtk-x11-2.0.so.0.2400.27)
==7897==    by 0x32D6DFCC29: wxEventLoop::Run() (in /usr/lib64/libwx_gtk2u_core-2.8.so.0.8.0)
==7897==    by 0x32D6E91B5A: wxAppBase::MainLoop() (in /usr/lib64/libwx_gtk2u_core-2.8.so.0.8.0)
==7897==    by 0xF80FFA8: wxPliApp::MainLoop() (app.h:195)
```

didli, 2015-05-06:
Strange : 1.2.6 binary is working, but only if i have NOT previously add an stl file on the plater tab. 
I will try the new commit in a few hours.

wolfmanjm, 2015-05-06:
FYI I am running with this version of  perl 5, version 14, subversion 3 (v5.14.3) built for i686-linux

wolfmanjm, 2015-05-06:
Sorry I am running perl 5, version 18, subversion 2 (v5.18.2) built for i686-linux-gnu-thread-multi-64int

alexrj, 2015-05-06:
That valgrind report basically means that two concurrent Perl threads are free()ing the same Perl variables at the time of their destruction. This shouldn't happen because all Perl variables are automatically cloned when a thread is spawned. I wonder how to debug this further...

didli, 2015-05-06:
My valgrind knowledge is quite close to 0 (-1 seems right too). Feel free to ask for any command you might think can help you. In the meantime, am i the only one reporting this issue ? Could it be a bad conf somewhere in my OS ? 
I don't want to monopolize your time for an issue I can workaround using terminal...
And also : I don't know if it's related but compiling from GIT in Fedora was only successful modifying this file : 
```/usr/share/perl5/vendor_perl/CPAN/Meta.pm``` line 60
```use Parse::CPAN::Meta 1.4414 ();``` -> ```use Parse::CPAN::Meta 1.4405 ();```

alexrj, 2015-05-22:
You're not the only one reporting this, but I'm afraid the bug lies somewhere outside Slic3r's code. Maybe it's inside perl itself, whose support for threads is not up to the job. The mid-term plan is to continue porting everything to C++/XS and do threading from there...

lordofhyphens, 2017-03-19:
This is probably fixed now, it just sliced for me (or it's not a bug in Slic3r at all and there's something dirty in your environment).

http://dl.slic3r.org/dev for fresh builds

Closing

closed 2017-03-19 by lordofhyphens