slic3r/Slic3r #2802: Slic3r with SLS or SLA

veneridesign, 2015-04-28:
Hi, I' trying to use Slic3r on a SLS machine, the only main problem is during  the filling operation, some times the scansion follow a stragegy that interrrupt a line , go in other places make other filling and than come back to finish, that interruption make big defects on the surface of sintered powder(that why the jump interrupt for to long time the process) 
Should be usefull have an option that simply "scan" the layer from top to down one layer and from left to right the next layer in order to minimize the interruption.

Thankyou

mypeopoly, 2016-12-05:
sorry to follow up on an one-year old topic but I am interested in this as well. was it solved? can I help?

lordofhyphens, 2016-12-05:
Don't worry about following up on old issues, this isn't a forum :)

If the issue is still open then odds are it wasn't solved. You are welcome
to look into the problem and document your findings.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mypeopoly, 2016-12-06:
@lordofhyphens Do you know of a slicer that does the most basic slicing without adjusting for any thing printer or machine specific? I can use that as a starting point.

alexrj, 2016-12-06:
@veneridesign, @mypeopoly, can you share more information about the machines you're using? 
Also, I don't fully understand the request. In the past I've been asked about transforming toolpaths to raster images so that SLS machines could perform linear horizontal scans by turning laser on/off very quickly for each pixel.
@mypeopoly, there's no such thing as "without adjusting" - a strategy is always needed. In this case we need to know the requirements better. Do you have videos or documentation you can share? Whenever a single continuous path is not possible because it has to be interrupted, how do you need it to be done?

mypeopoly, 2016-12-06:
@alexrj 
Hi, I am building my own laser based SLA printer that uses gcode as instructions to where and how fast should laser move.  I did most of my testing in Cura 15.04 which worked ok. Once I switched to Cura 2.3, the print quality actually dropped as I found out the Cura takes more "advanced" strategy when deciding toolpath. A lot of those strategies take consideration that FDM extruder moves slower and ooze too much in certain situation and to remedy, slicer would created toolpaths that are interrupted in the middle and rejoined later. (@veneridesign mentioned the same thing) Because laser doesn't have the same speed and oozing issues as FDM, the paths do not need to be interrupt.  Therefore a gcode slicer for SLA (and SLS) is best if it creates gcodes has the least interrupts.   

I hope that clarifies what we are looking to do and let me know if there is anything I can help.

alexrj, 2016-12-06:
I don't understand where you say:

> A lot of those strategies take consideration that FDM extruder moves slower and ooze too much in certain situation and to remedy, slicer would created toolpaths that are interrupted in the middle and rejoined later.

For FFF printing, optimal paths are continuous, with no interruptions and short travel moves. How is this different from other machines?

Please, can you post an example of a toolpath generated by Slic3r which you think is suboptimal for SLA? Let's discuss actual examples.