slic3r/Slic3r #2805: [Request] Layer Height in Print Settings - Advanced options

ML3D, 2015-04-29:
In addition to 'First layer height' it would be great to have a quick box entry for 'Top layer(s) height'.
With the option of selecting x number of top layers heights.

So I could select 3 top layers of .1mm wherever Slic3r is finishing off a top layer, while keeping the rest of the print at another height. This would really improve the finishing quality.

I can sort of do this in the layers feature, but it's more complicated and doesn't save as a setting for all prints. ( If a model has multiple top surfaces, I'd have to manually calculate and input the top layer measurement range, and this would then typically affect the whole print )

I would also really love the ability to change more than just layer height, especially the # of vertical shells, per layer range.

-------------

The ability of lift the z at the very end of the print, instead of wiping across the finished surface.

-------------

Finally, I've been doing top layer, surface finishing tests. The enemy of a fine quality surface finish on the top of a flat section is any stop/starts as well as right angle moves. I created a test object with rounded corners and I was able to increase the vertical shells to partially simulate a top fill pattern with a combination of curved and straight lines. However, the shells are not a continual extrusion, so there is the break that repeats. And the concentric infill doesn't fully match up with the edge and disrupts the smoothness. Otherwise, the outer sections are the smoothest overall top surface I've come up with. With all of that said, if a pattern could be generated that used a single extrusion, with a blend of straight and curved lines ( basically, no angles and as few stop/starts as possible ) I think we could achieve some incredibly improved top layer finishes on our prints.

[print_surface_test01](https://cloud.githubusercontent.com/assets/12176262/7402069/d32f8d40-ee7d-11e4-99e6-7e737dd803de.jpg)

Regards and thanks for all of the hard work you've done on Slic3r.

lordofhyphens, 2015-04-29:
Take a look at modifier meshes, you might be able to do what you want with those w/r/t more control over toolpaths. 

You can handle lift Z at end of print with custom gcode, just switch to relative mode with G91, issue G1 Zxxx, and go back to absolute with G90.

ML3D, 2015-04-29:
Thanks for the reply. I wondered if some gcode at the end might fix that, I'll try it out.

I do need to explore the modifier meshes a bit. But I still think having some ready to go input boxes/options would be very helpful and much faster. Plus, that way you can save your config for top surfaces.

alexrj, 2015-05-02:
I'm not sure about how thinner layer height improves quality of horizontal surfaces. Anyway, you might want to check this out, which is easier than modifier meshes: http://manual.slic3r.org/expert-mode/variable-layer-height

Thanks for the other hints and feedback.

closed 2015-05-02 by alexrj