slic3r/Slic3r #2807: layer heigh 0.2mm all pattern missing

yoyeung, 2015-04-30:
I find this different in v1.2.6.
Using this model: http://www.thingiverse.com/download:1169234
[image]
0.2mm per layer
[image]
0.3mm per layer

Nozzel size is 0.4mm.
is it a issue??

lordofhyphens, 2015-04-30:
Please post a copy of your config so we can see everything that is going 
on.

yoyeung, 2015-04-30:
here you are.

# generated by Slic3r 1.2.6 on Thu Apr 30 13:46:12 2015

[filament:Untitled]
bed_temperature = 0
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 3
extrusion_multiplier = 1
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.75
first_layer_bed_temperature = 0
first_layer_temperature = 200
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
slowdown_below_layer_time = 5
temperature = 200

[presets]
filament = Untitled.ini
print = Untitled.ini
printer = Untitled.ini

[print:Untitled]
avoid_crossing_perimeters = 0
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
default_acceleration = 0
dont_support_bridges = 1
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0
fill_angle = 45
fill_density = 20%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 0
first_layer_height = 100%
first_layer_speed = 30
gap_fill_speed = 20
gcode_comments = 0
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 80
interface_shells = 0
layer_height = 0.2
min_skirt_length = 0
notes = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 3
post_process = 
raft_layers = 0
resolution = 0
seam_position = random
skirt_distance = 6
skirt_height = 1
skirts = 3
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 20
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
thin_walls = 1
threads = 2
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 130
xy_size_compensation = 0

[printer:Untitled]
bed_shape = 0x0,200x0,200x200,0x200
before_layer_gcode = 
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
extruder_offset = 0x0
gcode_flavor = reprap
layer_gcode = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
pressure_advance = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 2
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
toolchange_gcode = 
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
z_offset = 0

[settings]
autocenter = 1
mode = expert

alexrj, 2015-12-19:
Thank you for reporting this. I fixed it in both the master and the stable branches.

closed 2015-12-19 by alexrj