slic3r/Slic3r #2808: Width smaller than the diameter of nozzle?

lonelymyp, 2015-04-30:
Print outer perimeter offset by half the width of the extrusion. Then half of the nozzle will be closed by the previous perimeter, this is equivalent to a decrease diameter of the nozzle.

something like this: 
[image]

alexrj, 2015-04-30:
But why?

lonelymyp, 2015-04-30:
After a few test prints, I came to the conclusion that the smaller the width the better it looks.
so I set the minimum allowable width of the extrusion to the perimeter.
I could not test the idea that the more thin perimeter will look even better =)

I'm no expert in printing, so maybe it's stupid =)

lordofhyphens, 2015-04-30:
Might be worth looking into, if only to attempt to improve inter-layer
adhesion. I'd be worried about the toolhead depositing plastic onto the
blocked section (as the previous layer is softened by the heat from the
toolhead), and you'd need to plan to reduce flow rate appropriately... and
you still have a hard limit of the nozzle width (as you need some perimeter
there to block).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-04-30:
I always stress that problems and goals should be declared first, then verified and discussed, and then finally it's time to move onto possible solutions.

Smaller extrusion width does look better for two reasons: 1) there's less plastic flowing at the sides, thus there's less chance of blobs caused by viscosity (and non-regular flow is less visible), and 2) the paths calculated by Slic3r for a thinner extrusion width follow the original object shape more accurately because the "tool-head" is smaller thus it can enter all the small features and curvatures.
This is why Slic3r uses a smaller extrusion width for the external loop (equal to the nozzle diameter, which is the minimum extrusion width).

The problem with your solution is that if the outer loop follows the inner one closely (like an offset) it will have no additional level of accuracy; if on the contrary we adjust it for greater accuracy we'll have a variable overlap with the inner loop, thus a not predictable effect on the outer shape.

Also I think that using the previous extrudate as a block does not work very well because nozzle partially melts it, and viscosity will generate blobs and other artifacts.

You can simulate that behavior by using two perimeters and tweaking the extrusion width settings and extrusion multiplier in Slic3r. The result won't be that good...

closed 2015-05-24 by lonelymyp