slic3r/Slic3r #2809: Gui advanced Extrusion width auto values improvement

robseee, 2015-04-30:
It would be helpful to have the user interface in the Print Setting -> Advanced -> "Extrusion width" frame to report the auto computed values between parenthesis () in real time.

Example implementation:
Default Extrusion width: [100%] mm or % (leave 0 for auto, *currently 0.4mm*)
and:
Default Extrusion width: [0.4] mm or % (leave 0 for auto, *currently 100%*)
and:
Default Extrusion width: [0] mm or % (leave 0 for auto, *currently 100%, 0.4mm*)

This would need to be implemented for any other option in the same tab (ie First layer, Perimeters, Infill, Solid infill, Top solid infill, Support material)

Thanks for you work.

braddo99, 2015-05-07:
In this area in general, the tooltip (and the behavior itself) is confusing. The tooltip for top solid infill for example says that you might want to have finer beads on the top surface and suggests a value of 95%, but then goes on to say that the width is computed over layer *height*. That would mean if you used 95% you would get an extrusion width of typically .95*0.2mm, which would never work. Why are you computing over layer height?

alexrj, 2015-05-22:
@braddo99, does it suggest a value of 95%? Where do you see that?

xoan, 2015-05-22:
It is not suggested nor recommended, only says _for example, 90%_.

https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L941

```cpp
Options["top_infill_extrusion_width"].tooltip = "Set this to a non-zero value to set a manual extrusion width for infill for top surfaces. You may want to use thinner extrudates to fill all narrow regions and get a smoother finish. If expressed as percentage (for example 90%) it will be computed over layer height.";
```

braddo99, 2015-06-06:
"You might want to" sounds like a suggestion to me. You're right it says 90% not 95. But, if you have a .4mm orifice and are using a 0.2mm layer height, this seems like it will attempt to make 0.2*0.9=.18mm line widths. Of course this will not work. I don't understand the reasoning behind making the advanced line widths a function of layer height. It screwed me up for a while as, although I knew it wasn't, I sort of forgot and was thinking it was a function of a fixed line width if set.  Severe under extrusion is all that happens and it was a head scratcher for quite a while for me. I would suggest this should be changed unless there's some logic behind being a function of layer height, and maybe some guidelines on why one might want that.

robse-git, 2023-09-23:
closing as too old.

closed 2023-09-23 by robse-git