slic3r/Slic3r #2810: Limit max. Extruder Flowrate/Feedrate without limiting Retract/Unretract Speeds

simonkuehling, 2015-04-30:
Hi,

i have trouble finding a suitable solution to limit the maximum volumetric extrusion rate of material that will ever be pushed through the nozzle for my extruders.

While the maximum feedrate of the extruder motor can be set in the firmware, it is not an adequate solution as it would also limit the retract and unretract speeds to the same level.
Thinking a little further, only the slicer can determine precisely which extruder move will most likely be actual extrusion, retract or unretract.

With features like "Combine Infill" and logic like "Gap Fill", there are quite some situations where extruders can reach their material flow limits by really wide or thick extrusion segments, despite already running with moderate overall feedrates.

To efficiently utilize the maximum print speed possible with a given printer, i propose to add a volumetric extrusion rate limit to the extruder configuration - so we do not have to artificially slow down general feedrates in order to account for some rare edge cases (like wide gap fill lines or really thick multi-layer infill paths)

(Or am i blind for an existing solution that already does just that?)

alexrj, 2015-05-03:
Have you considered M221?

https://github.com/alexrj/Slic3r/wiki/FAQ#q-how-can-i-enforce-a-volumetric-limit-for-extrusion

I'm not sure how it applies to retract and unretract. But shouldn't the limit apply to all moves, being normal print or retraction?

simonkuehling, 2015-05-03:
The M221 command is a proportional speed multiplier, decreasing or increasing all movements (including the corresponding extrusion of course). 
What i am searching for is an absolut flow-rate limit for the extruder - a value that is static for a specific hotend+nozzle+filament combination. The maximum feedrate the extuder can handle before slipping (mm/s or mm3/s).

It is quite easy to choose usable printing speeds for perimeters and infill that won't exceed the extruder flow limit - but it is quickly getting unpredictable with settings like 'combine infill' and special cases like wide gap fills.

Although the flow rate limit could be a machine-specific (=firmware) configuration, the firmware can not  easily distinguish between unretract (=fast!) and normal extrusion (needs to be limited), which is why i think it may be better fitting into the slic3r.

alexrj, 2015-05-03:
Ouch sorry, I wrote M221 but I was thinking about M203 as mentioned in the link...

simonkuehling, 2015-05-03:
Ah, yes - i just investigated M203 as well.. Unfortunately it is only available in Marlin, or at least not in Repetier Firmware... I am not completely sure if the functionality is exactly as i proposed, but i may well be.
Did you test M203 yourself in this regard?

lordofhyphens, 2015-05-03:
M203 is also available via smoothie hardware/firmware, as is hardware
retract  (G10/G11).

I am also wondering how temp and materials factor into your maximums.

whosawhatsis, 2015-05-03:
You're thinking of M220. M221 is a flow multiplier. It increases/decreases the amount of filament extruded relative to the distance moved along the other axes. It functions like Slic3r's extrusion multiplier.  

To answer your original question, I don't believe any of the firmwares have separate maximum speeds for the extruder during printing and during retract/recover, mostly because they don't really distinguish between the two. If any of them do have this ability, it's probably Smoothie.

This could, of course, also be a setting in the slicer (and I believe Simplify3D sets printing speeds in mm^3/s by default, though that has other problems because it ignores linear motion speed limits), but M220/M221 would still override it.

The ideal behavior would be for the slicer to take both a maximum linear speed (for printing moves, separate from travel speed, which can usually be higher) and a maximum volumetric speed, then dynamically adjust printing speed as it goes to avoid violating either of those limits, and also to avoid sudden changes in volumetric extrusion rate because the time it takes to re-equalize pressure can cause artifacts (particularly with a bowden). Then, the next step would be to come up with a way to dynamically adjust temperature relative to volumetric extrusion rate (only when that rate doesn't change suddenly) to allow fast extrusion of large parts without over-melting small features at the top, as so often happens.

simonkuehling, 2015-05-03:
You're right, i mixed up M221/M220.

Do you agree that your ideas can be split into the following three separate topics?

1) introducing a volumetric speed limit per extruder. (This is equal to what i had in mind - should be pretty straight forward to implement)

2) avoiding sudden changes in volumetric speed - now that we are watching this value anyway due to No.1 (good idea, but worth a separate issue i think)

3) dynamically adjusting temperature relative to volumetric extrusion rate. (Like No.2 a really good idea - but also worth another independent feature request in my opinion)

Anyway - looks a lot like the volumetric limit fits better into the slicer than into firmware logic..

whosawhatsis, 2015-05-04:
@simonkuehling agreed.

alexrj, 2015-05-22:
This is very interesting and doable.

So, we can add a *Max Print Speed* setting and allow to set any of the other current speed settings to 0/auto (this still allows to override the autospeed where wanted).

Then I would not make *Max Volumetric Speed* required, since it's not very user-friendly and not easy to calibrate. I would make it optional and default to calculating it as the maximum one given by the combination of *Max Print Speed* and the thinnest extrusion width used in the print. That would give us the upper speed limit, and any other thicker extrusion width will be handled by slowing the print speed.
What do you think?

simonkuehling, 2015-05-22:
Well, i'd actually prefer a dedicated *Max Volumetric Speed* parameter, which can more easily be provided as a machine-specific setting ("printer" settings) by e.g. a printer manufacturer with a config bundle. I think a combined calculation from *Max Print Speed* and extrusion widths may more likely be accidentally voided by the user when playing around with settings that one would not expect to affect the max volumetric speed limit...

Usually the *Max Volumetric Speed* limit can be aquired experimentally by manually extruding material with rising feedrates - and subtract a safety factor from the speed where the extruder started slipping.

alexrj, 2015-05-22:
I'm proposing to **have** that setting - but just not making it required. If user doesn't set it, I propose to calculate it the way I described before...

simonkuehling, 2015-05-22:
Sorry, my fault! I totally missed that part of your comment.

In this case i totally agree with your suggestion, may be a working default logic!

whosawhatsis, 2015-05-22:
Sounds good to me.

alexrj, 2015-05-31:
Okay, I implemented it.   Can you all please help testing? :)
Instructions:
* set *Max print speed* to whatever you like
* set *Max volumetric speed* to whatever you like or leave 0 for auto
* set all other print speeds to 0 (unless you still want to set some of them manually)

lordofhyphens, 2015-05-31:
I will give it a spin when I get my printer back up and running.

simonkuehling, 2015-06-02:
Sure we will!

After first tests in different scenarios there are two questions/issues i found so far:

* when leaving some of the dedicated speed settings activated (like perimeter speed), the *Max volumetric speed* isn't taken into account as a max limit - i would expect a "whatever feedrate is lower" behavior in order to never exceed the max volumetric limit whatever speed you choose. Is this intended or just not yet taken into consideration?

* in order to be compatible with multi-extruder setups i would suggest to move *Max volumetric speed* into the filament settings - so you can set a maximum extrusion rate for each material individually. This is also very very useful for printing flexible materials where we could just leave our print speed settings at their default values and only put a very low volumetric speed limit into the filament setting 

In general, these two aspects aid in the separation of mechanical machine characteristics from material specific characteristics i think. The print speed settings are often tuned to match the mechanical capabilities of the printer - limiting perimeter speeds for best surface quality and pushing the limits for infill to reduce print time. But they may well be clipped by material properties through the *Max volumetric speed* setting.
This way we can also slowly print TPU together with support material that can be fed much faster (TPU/HIPS seems to work quite well in first tests!)

alexrj, 2015-06-02:
Hi @simonkuehling. I agree with the first point - I just didn't take it into consideration as I thought the new autospeed logic would be best isolated from any other things, but I'll do as you propose.

Regarding the position of the *Max volumetric speed* option, I really don't know where it should go. It would even fit in the *Printer Settings > Extruder* section...

simonkuehling, 2015-06-02:
Another bug seems to have slipped in somewhere during GCODE export:
When setting all print speeds to zero there is a division by zero in Gcode.pm:

<code>Illegal division by zero at Slic3r/lib/Slic3r/GCode.pm line 330.</code>

[image]

Only <code>Top solid infill</code>, <code>Bridges</code> and <code>Support material</code> may be zero - all other speed parameters at "0" generate the same error...
It seems to be independant from the actual STL that is beeing sliced.

The exported config.ini is:
[config.ini](https://www.dropbox.com/s/ghr71xb9v3wrxef/config.ini?dl=1)

simonkuehling, 2015-06-02:
@alexrj, yes - at first sight i also immediately thought that it basically is a extruder-specific parameter. It was only by accident that we tested the new feature with really soft TPU filament yesterday when we had the idea that it could possibly better fit into the filament settings. The maximum volumetric speed is obviously not only dependant on the melting speed but also on the filament elasticity. A factor i missed till now.

Since the filament temperature settings (both melting temperature and bed temps) are mostly tuned for one specific 3D printer anyway, the filament and printer presets are already cross-linked in some way

alexrj, 2015-06-02:
@simonkuehling, that's true.

I just fixed the "Illegal division by zero" error and applied *Max Volumetric Speed* to any extrusion (not just to those using autospeed).

whosawhatsis, 2015-06-02:
Because it's a "use the lowest value" setting, there's no reason it couldn't appear in both printer and filament settings. You could have an extruder_max_volumetric_rate and a filament_max_volumetric_rate, or equivalent.

simonkuehling, 2015-06-02:
@whosawhatsis, good point...

smsagent, 2015-06-11:
Hello all!
Thanks for great work!! - this feature is great to see finally integrated into Slic3r directly. I have been using my own script for over a year, which did almost the same -> after Slic3r exported .GCO, it called my script, which wrapped retract moves with "M203 E25" and "M203 E5" to speed up/slow down extruder before/after retract. The slow down command slowed on the maximum extrusion speed my extruder should reach and firmware did all limitations in background.
I have two questions or may be advances:
1st - this Max Volumetric Speed is Extruder dependent, so if I have for example a small printer farm (which I have), I have more printers with different nozzle diameters etc. So it should be nice, to see this setting of Max Volumetric Speed at Extruder settings under Printer settings, not under Print settings.
I have printers with double extruders too and both dont have exactly the same speed. If I print with PLA and PVA as support, PVA prints slower, so I need to slow down the whole print.

2nd - when I tune my extruder, I use "G1 E100 Fxx" to extrude 100mm of filament to see, if stepper have enought power and dont loose any step. This speed of F-parameter is in mm/min, so I need to recalculate it to volumetric. I think, that more practical is to enter this Max Volumetric Speed as "Max Extrusion Speed (mm/min or mm/sec)".
This is my practical ideas :-) Should you think about them and make a little changes to next release please?
Best Regards!
Ondrej - REPRAPOBCHOD.CZ

smsagent, 2015-06-12:
Hello again!
I agree with @whosawhatsis!
Move that Max Volumetric Speed to each extruder section (I printer have more than one extruder, let me set separate values) and Max Volumetric Speed under Filament setting.
The lower value will be used, I think? ;-)
Regards!

kefir-, 2015-06-12:
I also agree. I just switched from PLA to ABS, and now I have to update the Print Settings instead of the Filament Settings. I also recently switched nozzle from 0.35mm to 0.5mm, and while I'm only experimenting with this, I assume that I can print at a higher volumetric speed with the bigger nozzle. Again I had to update this in the Print Settings, rather than under the Extruder settings.

alexrj, 2015-06-12:
I also agree with you - but I'd like to know whether the feature works and provides good results. I have done little testing so far. Can any of you provide feedback about the quality improvements with autospeed?

smsagent, 2015-06-12:
Max Volumetric Speed works great, it does exactly the same result I had my own script, so my word is, that this feature is OK, just move it under extruder setting.
The second feature - autospeed - I have not tested it so much, but I will start printing some larger print job on Monday, so I will test it as much as I can and give you my experiences here until the end of the week, OK?
Regards!

lordofhyphens, 2015-06-12:
@smsagent Except that the max flow rate is also tied to the material used
and temperature.

I'm not sure that moving it to "extruder" is the best fit.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

smsagent, 2015-06-13:
@lordofhyphens Yes, max flow rate depends on pressure flow loss, temperature and material used.
If we want to be definite, there must be material flow config under filament setting (dependent on material type and temperature) and extruder max pressure under extruder setting.

But lets keep it as easy, as possible for now and put this setting to the place, where user expects it most. My own experience is, that I use similar setting for both ABS and PLA (different temperature, but speed is almost the same). And because I have more printers with many nozzles, I expect to have Extruder dependent setting, otherwise, I will need filament setting for each printer and filament used - too many configs!

So my own opinion is to have Max Volumetric Speed setting under extruder and in the future, we can think about some modificator value under filament setting, because if I use some flex filament, I realy need to slow down, so under filament setting shloud be Max Volumetric Speed Ratio.

Regards!

simonkuehling, 2015-06-16:
@smsagent thanks for the precise summary.

I like the idea of a proportional modificator factor under filament settings complementary to the extruder specific *Max Volumetric Speed* setting.
This way both are directly connected with eachother, which emulates the underlying physical effects quite closely (or as close as we get with empirical parameters as an input)

@alexrj, right now i only quite extensively tested regarding print failures due to too high extrusion rates as this was the first priority in my cases. For this i can confirm huge success already, it works remarkably well and make slic3r profiles much easier to modify without knowing the machine limits so well.
I will issue some performance tests for surface quality impacts over the next days as well - especially for the case of thinner outer perimeters..

xoan, 2015-06-16:
I'd started to try this cool autospeed feature in 1.2.8-experimental. Nice addition.

I've been searching but seems to be not reported yet: for the first layer speed configured as a percentage, full autospeed is used in speeds configured as 0 instead of lowering this autospeed to the first layer speed percentage.

josla972, 2015-06-24:
I really like this feature. I have been experimenting with similar processing, but my focus was rather to rate-limit the changes in volumetric extrusion speed rather than keeping it constant (i.e., honoring the user configured speed but rate limiting the changes in speed). Somewhere along the way I gave up (because of lack of time, perl skills and lack of insight in the Slic3r code, not because the idea would not work) and switched back from bowden to direct extruder. I guess one would need good time estimates of the time spent for each different extrusion speed for the rate limitation to be meaningful. The nice thing about this would be that we could allow the pressure to increase or decrease slowly, thus having sufficient time to reach the correct pressure before we begin printing at a different speed. This optimization problem should be solved such that we only punish higher extrusion speeds by slowly decreasing them before attempting to print with a slow extrusion speed (since it is generally not bad for the print quality if we print slowly, but ramping up speed on a part of the layer that should have been printed slowly can be devastating).

Hexman64, 2015-06-25:
I set all the extrusion widths to 0, and as long as the speed settings are above 0, I get a nice gap fill which I call "the new style":
[image]

When I then change the speed settings, especially the gap fill speed setting to 0, gap filling is done the legacy way, which would give a badly shivering printer, especially when it tries to keep a (high) constant volumetric extrusion:
[image]
(by the way, for this model you see a gap fill error in one of the lower layers)

I dislike the shivering gap fill, but would need constant volumetric speed when printing Ninjaflex material. Am I doing something wrong? Slic3r is Windows 1.2.9 x64

smsagent, 2015-06-30:
Hello! So I have been testing and testing.. now I think, I have found a little issue - "Max Volumetric Speed" does not calculate with "Extrusion multiplier"? I have generated two G-Codes of the same object, one setting have Extrusion multiplier set to 1 and the second setting is 1.1. Now when I compare these two G-Code files, I see no speed changes on F parameters.
I have fine-tuned my extruder max speed, so this small increase of extrusion multiplier is very noticeable on quality, so I have found this issue.

teppix, 2015-07-03:
+1 for fixing the issue mentioned by @Hexman64 

I think auto volumetric flow is more important for gap fill than it is for any the other speed settings. Since extrusion widths for gap fill can vary quite a bit, it isn't optimal to set this as a linear speed.

alexrj, 2015-07-03:
@Hexman64, that's off-topic here. Can you please move that into a new issue so that we can discuss the issue?

alexrj, 2017-03-04:
@smsagent, Extrusion Multiplier is only applied to the extrusion values in order to correct the error between the volumes calculated by Slic3r and the actual volumes. It's a fudge factor to fix the error introduced in Filament Diameter and/or your firmware E steps. Applying that to Max Volumetric Speed would introduce a recursive error in logic. :)

closed 2017-03-05 by alexrj