slic3r/Slic3r #2811: Support 3MF specification

harriv, 2015-05-01:
It's a brand new specification: http://3mf.io/what-is-3mf/3mf-specification/

alexrj, 2015-05-01:
Yeah, saw that. But I couldn't find any example file. The specs are not that clear, despite the presence of the XSD. Can you find an example file?

harriv, 2015-05-01:
I'll look. It may take some time before there's clear examples around..

harriv, 2015-05-02:
Windows 3D Printing SDK should has have sample files: https://msdn.microsoft.com/en-us/windows/hardware/bg183398.aspx

Unfortunately it only install on Windows 8+, which I don't have.

drayde, 2015-05-18:
Still need an example 3mf file? Email me at mail-at-andreaskahler.com

lonelymyp, 2015-05-24:
sample mug:
3mf: https://drive.google.com/file/d/0B2sGop41unyOamt0aWJZMEVyMUU/view?usp=sharing
stl: https://drive.google.com/file/d/0B2sGop41unyOQ0c4aHFseER4QUU/view?usp=sharing
[image]

rogersachan, 2017-01-21:
@alexrj The whole thing is now open source https://github.com/3MFConsortium and is supported by Cuda.

rogersachan, 2017-01-21:
Their documentation is noticeably better now, and is implemented and a list of implementations are here: http://3mf.io/adoption/ .

lordofhyphens, 2017-01-21:
Yeah, I have been keeping an eye on lib3mf, just haven't had the time to
dive in and implement a link to our internal formats.

rogersachan, 2018-05-09:
Is this issue resolved as of #4046 ?

lordofhyphens, 2018-05-09:
Yes, it should be.

closed 2018-05-09 by lordofhyphens