slic3r/Slic3r #2812: Over extrusion at Start of External perimeters (Slic3r: 1.2.6)

LAZLOWcv, 2015-05-01:
Slic3r:1.2.6
Printer:CoreXY
Settings: 3 Perimeters and Random Seaming.

I looked Through the issues and didn’t see this, If it was I am Sorry for wasting your guys time.
The issue I have found occurs at the start of the external perimeter, the extruder slightly shifts when switching from the previous perimeter to the external. It is at this point when the over extrusion occurs and the layer is slightly bulged out beyond the previous layer for approx 5-10mm before correcting itself and smoothing out. When this occurs it causes many little ridges in the print. (Left Part in Picture)
Also I am using 3 perimeters and random start points.
I have been having this problem for a couple of months now and was thinking it was moisture in my filament but recently i discovered that this issue does NOT Occur in Slic3r 1.1.7.
Using the exact Same settings in both versions I get the results pictured below. Slic3r 1.2.6 on the left and 1.1.7 Stable on the Right. I did not try the other Seaming Settings so i am not sure if this is only related to the random setting or not. 
Not sure if anyone else has seen this issue but i wanted to get it out there, as it can make the prints look quite bad.

Thanks for your guys time.

[image]

lordofhyphens, 2015-05-01:
Please consult the contribution guidelines and provide a copy of a print
stl (linked) that shows the problem and a copy of the slic3r configuration
that you used.

alexrj, 2015-05-02:
Yes, please, see the guidelines and provide a reproducible test case. Please make sure the issue is visible in the G-code preview (either inside Slic3r itself or with another preview tool like gcode.ws).

LAZLOWcv, 2015-05-03:
Link below to Config Files and STL pictured above. I want to mention again this is not STL Dependent, It occurs on every STL i have printed with the 1.2.6 version.
Exactly the same config settings used on 1.1.7 and the problem is gone.
Also it Does NOT show on any Gcode Viewers that i have noticed. the location where the issue occurs is marked with a red arrow. see pics.
Thanks

[image]
[image]

https://drive.google.com/folderview?id=0B8oFTvTHSWISfk5BOWxyS2ktbUp4Mm51Tkk3MnJDcDl1SjRjMHZTZFloRVJ6VVRDRDBJa3M&usp=sharing

alexrj, 2015-05-04:
I don't understand what the problem is. If it's not visible in the G-code, then it's generally not a Slic3r problem. You might want to experiment with the Seam Position option.

LAZLOWcv, 2015-05-04:
I agree with you opinion to an extent, but how can that explain 1.1.7 having a perfect print and 1.2.6 having this issue with the same exact settings on both slic3rs? I will do a little more research for you ASAP. I can generate gcode from both and compare the positions of the axis and extruder values, maybe that will yield a result. I would like to note that I have had this problem on two different printers. I'll also try another seam setting as well.  I'll update you on my findings. 
Thanks.

simonkuehling, 2015-05-04:
One major difference between 1.1.7 and 1.2.6 is that now by default Slic3r uses a thinner outer perimeter width than the other perimeter loops - to aid in surface quality and detail resolution.
In 1.1.7 all perimeter loops had equal width.
My guess here is that this introduces a "jump" in material flow rate (from higher to lower) between the second-last and last perimeter loop. As the nozzle needs some time until the inner pressure settles at the lower state again, there is some over-extrudate visible on the first few milimeters of the outer loop.

This could be confirmed by statically enforcing equal perimeter widths for all loops in 1.2.6 (print settings -> advanced)... I am curious for any results!

LAZLOWcv, 2015-05-04:
Simon
Your comment makes sense and I suspect your right on the money. I'll try this on my printer tomorrow and update you guys. 
Thanks.

LAZLOWcv, 2015-05-05:
Simon,
Looks like you where correct. I just tried a print with your suggestion and it looks perfect.
I set all of the extrusion widths to 0.4 and didnt have one line in the print.
Whether this is considered a bug or not its good for this issue to be out there for anyone else with this problem. Then again maybe i am the only person out there that has had this issue, not to mention on two seperate printers. 
Thanks

simonkuehling, 2015-05-05:
Thanks for your quick confirmation - i have observed this effect on several prints as well (with ABS, 2.85mm filament, 0.35 mm nozzle and even with modest max speeds of only 30mm/s)

Another issue with the thin outer perimeter is that the maximum printable overhang angle is clearly reduced quite a bit (especially for warping-prone materials like ABS, PC, PA) since the overlap with the perimeter below is smaller.

@alxrj, maybe the outer-perimeter-width behavior could be deactivatable in some way (e.g. by setting a width value of "-1" or adding a checkbox...)?
Then users affected by this can still profit from auto-generated extrusion widths but opt to avoid the artefacts introduced with a thin outer perimeter...

LAZLOWcv, 2015-05-05:
Yep I agree an option to turn this on and off would be nice and it would allow for slic3r to handle the widths. Not sure if its possible but another solution that i thought of would be to start the thinner outer perimeter width earlier, slightly before the extruder shifts to the outside perimeter. That would allow for nozzle pressure to drop before the start of the external perimeter. Not sure if this would cause any side effects like small gaps or not but it was just a thought.
 Would be much easier to just turn it on and off though.

simonkuehling, 2015-05-05:
@LAZLOWcv, since the pressure drop is caused by the change of material flow rate in the nozzle, an easier way to avoid huge pressure jumps here would be to gradually adjust printing feedrates (gradually slowing down before reaching the first outer perimeter point) - instead of using dynamic perimeter widths.
In fact this idea also came up in issue #2810 just yesterday from @whosawhatsis - i think your example here was exactly what he had in mind...

lonelymyp, 2015-05-24:
The problem here is similar to this?
[image]

simonkuehling, 2015-05-24:
@lonelymyp, yes - that's likely the issue discussed here.

kefir-, 2015-05-24:
I've also had a lot of artifacts like this when printing with recent slic3r dev, but I thought it was something to do with the filament. 

I understand a workaround is to set the same extrusion width for all sections, but it'd be interesting to also try to keep the pressure equal by speeding up the thinner external perimeter. I'm not able to test for a few days as least, does anyone else want to try to verify this?

lonelymyp, 2015-05-24:
I guess there is a delay when moving to a new perimeter. nozzle moves slowly, and the pressure increases rapidly.

alexrj, 2017-03-04:
Autospeed fixes this in my opinion, so I'm closing it.

closed 2017-03-04 by alexrj