slic3r/Slic3r #2813: Version 1.2.6 - consistent crash in Windows 7 64-bit

WZ9V, 2015-05-03:
If I load the high definition bonsai planter model from thingiverse, I can crash slic3r consistently by enabling supports and scaling to 225%

Without supports it works fine at 225%

This is the model  http://www.thingiverse.com/thing:592497

alexrj, 2015-06-02:
I tested on 1.2.8-dev and it doesn't crash.
Would you be so kind to test on the latest binary (1.2.7)?  I think it's fixed in that version. Let me know!

WZ9V, 2015-06-02:
Sure, tonight I'll download 1.2.7 and try the files that were causing problems in 1.2.6

WZ9V, 2015-06-03:
1.2.7 seems to get farther along but never seems to finish the generating support step.  I'll try it again when 1.2.8 is ready.

a4jp-com, 2015-06-09:
I just updated 1.2.8-dev and could load this file perfectly. It takes a while to load but no problems at all. WZ9V what is the support step problem?

WZ9V, 2015-06-09:
Not sure.  It just never completed and the app went unresponsive.  I tried twice both times I let it go for more than an hour and the progress bar just stayed in the same place.  I'll try 1.2.8 when it moves to experimental.

alexrj, 2015-06-20:
@WZ9V can you try 1.2.9? Thank you

WZ9V, 2015-11-12:
Sorry it took so long to get back to you, major printer issues.  Tested 1.2.9 running on Linux Mint 17.2 and it still hangs on the generating support step with that object.

a4jp-com, 2015-11-12:
How about 1.2.10-dev?

WZ9V, 2015-11-12:
I'll try and figure out how to set that one up and give it a try.

a4jp-com, 2015-11-12:
Download the files then go to the command prompt window. If perl is installed you can type in 

perl Build.PL --v

then after that 

perl Build.Pl --gui

WZ9V, 2015-11-13:
I was able to scale model to 250% and generate supports using 1.2.10-dev once I got it built.  I have generated the g-code file but have not tried to print it yet due to issues with my first layer again.  Once I get the first layer on my printer sorted out I'll try another print.  I don't have the Windows 7 machine anymore to try it there but it is working on Linux Mint 17.2 where it was still having problems when I tried it on that.  I think it is working now.  Once I get a print going I'll update with anything I see there but it looks good in preview and layer view.

a4jp-com, 2015-11-13:
Nice ^^

closed 2016-03-18 by alexrj