slic3r/Slic3r #2814: Missing raft on thick layers

Cylindric, 2015-05-04:
I'm just experimenting with a new 1.2mm nozzle, and have noticed that no raft is being created when layer height is 0.9mm or more.

I am using Slic3r 1.2.6, with a simple 20mm cube and all the (default) profiles. All I changed for the test is:

* Nozzle set to 1.2mm
* Layer height set to 0.9mm
* First layer height set to 0.9mm
* Raft set to 1

Preview - you can see the gap under the object:
[image]

Layer 1: 0.9mm - can only see the skirt:
[image]

Layer 2: 2.3mm - straight to the object over 2mm above the bed:
[image]

Cylindric, 2015-05-04:
Forgot to add, here's a copy of the entire config: https://gist.github.com/Cylindric/1b36d2955806b07ffc2d

alexrj, 2015-06-02:
Fixed, thank you.

closed 2015-06-02 by alexrj