slic3r/Slic3r #2815: IO::stringy

a4jp-com, 2015-05-05:
IO::stringy says to use --force to install but it doesn't work. Even when --force is being used, it still has the same error message. I'm just trying to install the newest ZIP file on a Windows 7 64-bit PC.

https://github.com/alexrj/Slic3r/archive/master.zip

alexrj, 2015-05-05:
Can you please provide a full report, including a full transcript of your terminal session showing the commands you typed and the errors you got?

c-born, 2015-05-05:
I just had the same problem when building from Git head d998d977543d45ecd17c401055139847516b532a 
I don't know why it decided to download IO-stringy-2.111 and build it, never has before.
I found this https://rt.cpan.org/Public/Bug/Display.html?id=103895 which seems to describe the same problem, although when building on Windows 8.1. Seems the problem is tests failing due to some incorrect offsets in the results, although the build is actually ok.

I opened my Perl build command prompt in the directory where the build failed (in my case C:\Users\cborn\.cpanm\work\1430807375.13512\IO-stringy-2.111) and build it manually:

dmake
dmake test   (fails)
dmake install

After that the standard perl build succeeded.

(Although the resulting Slic3r build has another bug - if you open the "About" dialog there seems to be no way to close it again!)

alexrj, 2015-05-05:
Unfortunately, without seeing the actual errors I can't really help :-(

c-born, 2015-05-05:
@alexrj I've just worked through the code, and the errors in the test cases are due to the hard-coded offsets in IO_InnerFile.t, exactly as described in the bug report in https://rt.cpan.org/Public/Bug/Display.html?id=103895, so it appears to be a reported upstream bug.

FYI My results in running the make test are:

```
C:\Users\cborn\.cpanm\work\1430807375.13512\IO-stringy-2.111>dmake test
"C:\dev\CitrusPerl\CitrusPerl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_h
arness(0, 'blib\lib', 'blib\arch')" t/*.t
t/IO_InnerFile.t .... Failed 7/7 subtests
t/IO_Lines.t ........ ok
t/IO_Scalar.t ....... ok
t/IO_ScalarArray.t .. ok
t/IO_WrapTie.t ...... ok
t/simple.t .......... ok
t/two.t ............. ok

Test Summary Report
-------------------
t/IO_InnerFile.t  (Wstat: 0 Tests: 7 Failed: 7)
  Failed tests:  1-7
Files=7, Tests=125,  1 wallclock secs ( 0.05 usr +  0.05 sys =  0.09 CPU)
Result: FAIL
Failed 1/7 test programs. 7/125 subtests failed.
dmake:  Error code 255, while making 'test_dynamic'

```

I haven't tried using --force so I don't know if that works or not - I didn't need to as my work-around as described above fixed it for me, and hopefully I won't see the problem again until they release IO-stringy-2.112, if ever!

markwal, 2015-05-05:
This is Windows only.  dskoll maintainer of IO::stringy added a new test in 2.111.  Windows CR/LF's show up in the stream breaking the test's expectations.  Nothing has changed in the functionality with respect to this, just the test, so IO::stringy will work on Windows as well as it did before.  --force worked for me by ignoring the test result and installing it anyway and then build.pl was satisfied the next time I ran it.

a4jp-com, 2015-05-05:
I just updated 20 minutes ago and maybe the IO:Stringy problem has been fixed but there is an IO:Scalar error instead.

perl Build.PL -v

then 

perl Build --force

is what I typed before.

After that perl Build.PL --gui

Here an image for the problem now.

[image]

[image]

markwal, 2015-05-06:
The --force isn't for build.pl it's for cpanm:
cpanm --force IO::Scalar

cpanm installs IO-stringy-2.111 as part of installing IO::Scalar
Do that cpanm --force above and then do perl Build.PL

markwal, 2015-05-06:
```
>cpanm --force IO::Scalar
--> Working on IO::Scalar
Fetching http://www.cpan.org/authors/id/D/DS/DSKOLL/IO-stringy-2.111.tar.gz ... OK
Configuring IO-stringy-2.111 ... OK
Building and testing IO-stringy-2.111 ... FAIL
! Testing IO-stringy-2.111 failed but installing it anyway.
Successfully reinstalled IO-stringy-2.111
1 distribution installed

>perl Build.PL
...
IO::Scalar is up to date. (2.111)
...
```

a4jp-com, 2015-05-08:
Thank you Markwal. I got the same error. Are they both broken?

markwal, 2015-05-08:
Yes, but not in a way that matters once you do --force.  It says FAIL, but with force it goes ahead and installs anyway and then you should be able to run Build.PL and Slic3r fine.

a4jp-com, 2015-05-08:
Okay. Thanks for the advice.

a4jp-com, 2015-05-08:
[image]

markwal, 2015-05-08:
cpanminus and lwp are different failures we'll have to look at the build.log's they reference to figure otu why. App::cpanminus worked for me, but I might have updated it elsewhere before running Build.PL. You could try installing separately (cpanm App::cpanminus)

Doing cpanm LWP::UserAgent definitely fails that way for me as well.  In it's case it's also because 27 tests failed.  cpanm --force LWP::UserAgent will do the same thing for it as IO::Scalar to get you up and running, but someone should make sure that the LWP issue has been logged with ETHER.  You could.  Either way, not really a bug in Slic3r.

But also, before you do Build.PL --gui you need to do Build.PL by itself.
cpanm --force IO::Scalar
perl Build.PL
perl Build.PL --gui

gege2b, 2015-05-11:
I got the same error on LWP:UserAgent (and Stringly/Scalar too, but it's OK now)

And it seems to screw up my citrus install (x64 on win7), because right after this error, it ALWAYS fail to retrieve packages (saying that "cpanmdb")

Sorry for this cheap description, I'm not on my home PC right now

Well, it's definetly not a slic3r error for me, but maybe someone can help ?

alexrj, 2015-05-17:
@gege2b, that might be a network/mirror issue as well. In order to troubleshoot it you might want to type `cpanm -v App::cpanminus` and check the output. (Are you running citrusterm.bat by the way?)

gege2b, 2015-05-18:
Hi Alessandro

Yes, I'm running citrusterm :)

Here's the output : 

    C:\citrusperl>cpanm -v App::cpanminus
    cpanm (App::cpanminus) 1.7034 on perl 5.016003 built for MSWin32-x64-multi-thread
    Work directory is C:\Users\XXX/.cpanm/work/1431942314.6568
    You have make C:\citrusperl\mingw64\bin\dmake.exe
    You have LWP 6.13
    Falling back to Archive::Tar 1.82
    Searching App::cpanminus () on cpanmetadb ...
    ! Finding App::cpanminus on cpanmetadb failed.
    Searching App::cpanminus () on metacpan ...
    ! Could not find a release matching App::cpanminus () on MetaCPAN.
    Searching App::cpanminus on mirror http://www.cpan.org ...
    Downloading index file http://www.cpan.org/modules/02packages.details.txt.gz ...
    Uncompressing index file...
    ! cannot open file 'C:\Users\XXX/.cpanm/sources/http%www.cpan.org/02packages.details.txt.gz': No such file or directory opening compressed index
    ! Couldn't find module or a distribution App::cpanminus

I get this error as soon as LWP:UserAgent was downloaded (giving me  a lot of errors), and then I just can't download any package using cpanm (it works using cpan )

alexrj, 2015-05-18:
Maybe you have a broken LWP::UserAgent installation. You could try to reinstall it using the `cpan` command and investigating any error should it throw...
(`cpanm` relies on LWP::UserAgent in order to download the packages)

gege2b, 2015-05-18:
that's what I've done, here's the output : 

    C:\citrusperl>cpan -i LWP::UserAgent
    Reading 'C:\citrusperl\cpan\Metadata'
      Database was generated on Wed, 13 May 2015 04:41:02 GMT
    Fetching with LWP:
    http://www.perl.org/CPAN/authors/01mailrc.txt.gz
    LWP failed with code[500] message[write failed: Bad file descriptor]
    Warning: no success downloading 'C:\citrusperl\cpan\sources\authors\01mailrc.txt.gz.tmp5632'. Giving up on it.
    Fetching with LWP:
    ftp://ftp.perl.org/pub/CPAN/authors/01mailrc.txt.gz
    Reading 'C:\citrusperl\cpan\sources\authors\01mailrc.txt.gz'
    ............................................................................DONE
    Fetching with LWP:
    http://www.perl.org/CPAN/modules/02packages.details.txt.gz
    LWP failed with code[500] message[write failed: Bad file descriptor]
    Fetching with LWP:
    ftp://ftp.perl.org/pub/CPAN/modules/02packages.details.txt.gz
    Reading 'C:\citrusperl\cpan\sources\modules\02packages.details.txt.gz'
      Database was generated on Mon, 18 May 2015 14:29:02 GMT
    ..............
      New CPAN.pm version (v2.10) available.
      [Currently running version is v1.9800]
      You might want to try
        install CPAN
        reload cpan
      to both upgrade CPAN.pm and run the new version without leaving
      the current session.
    
    
    ..............................................................DONE
    Fetching with LWP:
    http://www.perl.org/CPAN/modules/03modlist.data.gz
    LWP failed with code[500] message[write failed: Bad file descriptor]
    Fetching with LWP:
    ftp://ftp.perl.org/pub/CPAN/modules/03modlist.data.gz
    Reading 'C:\citrusperl\cpan\sources\modules\03modlist.data.gz'
    DONE
    Writing C:\citrusperl\cpan\Metadata
    LWP::UserAgent is up to date (6.13).

If I try with "cpan -if", then I get a ton of errors like : 

    #   Failed test at t/local/http.t line 129.
    #                   'Can't connect to portable7:61718 (10061)
    #
    # LWP::Protocol::http::Socket: connect: 10061 at C:\citrusperl\cpan\build\libwww-perl-6.13-NxWFKn\blib\lib/LWP/Protocol/http.pm line 47, <DAEMON> line 1.
    # '
    #     doesn't match '(?^m:^X-Foo:\s*Bar\n)'

and then ...

    # Looks like you planned 63 tests but ran 36.
    # Looks like you failed 27 tests of 36 run.
    # Looks like your test exited with -1 just after 36.
    HTTP Server terminated
    t/local/http.t .......... Dubious, test returned 255 (wstat 65280, 0xff00)
    Failed 54/63 subtests
    t/local/protosub.t ...... ok
    t/net/http-get.t ........ skipped: no net config file
    t/net/http-post.t ....... skipped: no net config file
    t/net/http-timeout.t .... skipped: no net config file
    t/net/mirror.t .......... skipped: no net config file
    t/net/moved.t ........... skipped: no net config file
    t/net/proxy.t ........... skipped: no net config file
    t/robot/ua-get.t ........ 2/8 HTTP Server terminated
    t/robot/ua-get.t ........ Failed 5/8 subtests
    t/robot/ua.t ............ 2/7 HTTP Server terminated
    t/robot/ua.t ............ Failed 5/7 subtests
    
    Test Summary Report
    -------------------
    t/local/http.t        (Wstat: 65280 Tests: 36 Failed: 27)
      Failed tests:  2-20, 22, 25, 27, 29-30, 33-35
      Non-zero exit status: 255
      Parse errors: Bad plan.  You planned 63 tests but ran 36.
    t/robot/ua-get.t      (Wstat: 0 Tests: 8 Failed: 5)
      Failed tests:  1-3, 5, 7
    t/robot/ua.t          (Wstat: 0 Tests: 7 Failed: 5)
      Failed tests:  1-3, 5, 7
    Files=16, Tests=104, 43 wallclock secs ( 0.17 usr +  0.11 sys =  0.28 CPU)
    Result: FAIL
    Failed 3/16 test programs. 37/104 subtests failed.
    dmake.exe:  Error code 255, while making 'test_dynamic'
      ETHER/libwww-perl-6.13.tar.gz
      dmake.exe test -- NOT OK
    //hint// to see the cpan-testers results for installing this module, try:
      reports ETHER/libwww-perl-6.13.tar.gz
    Running make install
    Appending installation info to C:\citrusperl\lib/perllocal.pod
      ETHER/libwww-perl-6.13.tar.gz
      dmake.exe install  -- OK

It seems like he tries to connect to localhost, but without success.
I checked the **great** windows firewall, but it's OK

but as long as this isn't a slic3r bug, I don't want to bother you more about that

alexrj, 2017-03-18:
The IO::Scalar issue is an upstream issue. I'm now using `--notest` (same as `--force`) in Build.PL now.

closed 2017-03-18 by alexrj