slic3r/Slic3r #2816: Bug 1.2.7-dev: slic3r fails to run

Fat-Zer, 2015-05-05:
slic3r fails to run with following message:
```
$ ./slic3r.pl
Bareword "X" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 53.
Bareword "X" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 53.
Bareword "Y" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 54.
Bareword "Y" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 54.
Bareword "X" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 54.
Bareword "X" not allowed while "strict subs" in use at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 54.
BEGIN not safe after errors--compilation aborted at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/Geometry.pm line 513.
Compilation failed in require at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/ExPolygon.pm line 8.
BEGIN failed--compilation aborted at /home/alexander/projects/reprap/Slic3r/lib/Slic3r/ExPolygon.pm line 8.
Compilation failed in require at /home/alexander/projects/reprap/Slic3r/lib/Slic3r.pm line 39.
BEGIN failed--compilation aborted at /home/alexander/projects/reprap/Slic3r/lib/Slic3r.pm line 39.
Compilation failed in require at ./slic3r.pl line 15.
BEGIN failed--compilation aborted at ./slic3r.pl line 15.
```
Perl version: 5.20.2

I suppose it was introduced with commit 5eb3bc5 which removed those constants.

alexrj, 2015-05-05:
Have you executed Build.PL?

Fat-Zer, 2015-05-05:
> Have you executed Build.PL?

yes

And readding X, Y and Z constants to the Slic3r/lib/Slic3r/Geometry.pm fixes the issue.

alexrj, 2015-05-05:
I think you haven't actually run Build.PL or it didn't succeed compiling. I confirm current git HEAD compiles and runs fine.

Fat-Zer, 2015-05-05:
yep... You were right... the Build.PL failed.

closed 2015-05-05 by Fat-Zer

munga4, 2016-05-31:
Am trying on mine and am having the same issue.

lordofhyphens, 2016-05-31:
On 1.3.0-dev?