slic3r/Slic3r #2818: Request: Avoid extrusions starting at overhangs

dapoofyhairdude, 2015-05-05:
I'm encountering an overhang issue that I believe can be fixed with a setting that causes slicing to avoid starting extrusions at overhangs. Please take a look at this photo:
http://i.imgur.com/9EaF7pm.jpg

Right below the nozzle are two overhangs that should be mirror images of each other. The right side is sliced such that the extrusion starts on the right (away from the overhang). The left side is sliced such that the extrusion starts on the right (on top of the overhang). This is causing way more curling/drooping on the left side. I know my printer is capable because the right side is near perfect.

I realize changing seams to random may fix this, but I prefer my seams to be aligned.

alexrj, 2015-05-06:
Actually Slic3r should already do this.   Can you please read the guidelines and provide a full report in order to investigate the issue?
https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

xoan, 2015-05-07:
I can confirm this with [`y-corners.stl`][STL01] and the `- default -` config.

Only the external perimeters in layers just before bridging the holes are started in a non-overhang corner.

I guess the overhang detection algorithm is only for angles > 45º, but the main problem is that even in an overhang < 45º, if the steam is allways in the same corner, this corner always tends to curling up.

[STL01]: https://github.com/josefprusa/Prusa3-vanilla/blob/master/distribution/y-corners.stl

alexrj, 2017-05-02:
[15:22:08] <VanessaE>	 you should be able to use this STL to reproduce it, http://www.thingiverse.com/thing:1378604

bubnikv, 2017-05-12:
> I guess the overhang detection algorithm is only for angles > 45º, but the main problem is that even in an overhang < 45º, if the steam is allways in the same corner, this corner always tends to curling up.

The G-code generator of the prusa3d fork implements a weighting scheme for placing seams outside of the overhangs. This works differently than the origional code by @alexr. In my implementation, the overhang distance is calculated precisely for each perimeter point and this value is then used as a penalty for the seam selection.

VanessaE, 2017-05-12:
@bubnikv I think you mean @alexrj :smile: