slic3r/Slic3r #2820: Theoritical Background

sajis997, 2015-05-07:
Hello 

I am just going through the theories behind the idea of 3D printing. Is the source based on "Selective Laser Sintering" and its relative improvements ? 

Some references to the algorithm implemented in this library would be very helpful .

Is there any forum where issues related to this library can be discussed ?

Thanks

lordofhyphens, 2015-05-08:
First, your question is inappropriate for this forum  (a bug tracker for
slic3r). Kindly do not post here again unless you have a problem with or
feature request for the slic3r program.

Second, I suggest searching through the wiki at reprap.org and a general
Google search.

alexrj, 2015-05-13:
Yes, and if you want to see how Slic3r works just have a look at the source code. That's what open source is for ;-)

closed 2015-05-13 by alexrj