slic3r/Slic3r #2821: 1.2.7-dev: Interface pattern spacing = 0 not always respected for topmost interface layer

simonkuehling, 2015-05-08:
I've found a special case with the support material interface layers, where the topmost interface layer is not always generated with zero spacing when <code>Interface pattern spacing = 0</code> is set:

[image]

This seems like an edge case in the code since <code>Interface pattern spacing = 0.01</code> works well as a workaround at the moment:

[image]

Full model for reference (STL/config below)

[image]

[config_0 interface spacing.ini](https://www.dropbox.com/s/ja4fvalzbrpd5tc/config_0%20interface%20spacing.ini?dl=1)
[config_0.01 interface spacing.ini](https://www.dropbox.com/s/a5mrxnvd0994z0n/config_0.01%20interface%20spacing.ini?dl=1)
[bucke.stl](https://www.dropbox.com/s/56c1ovj6u6pxkx6/buckle_plated.stl?dl=1)
Slic3r 1.2.7-dev from latest git

valentt, 2017-03-30:
This is an olfd bug, has it been fixed in recent releases?

lordofhyphens, 2017-03-30:
Good question. You can download his test case and see ;-)