slic3r/Slic3r #2823: Portable Slic3r - enhancement request

braddo99, 2015-05-09:
I have a central (dropbox) installation of lots of tools which I access and use from various PCs on my network. However, for Windows, Slic3r puts ini files etc into User/Roaming/Slic3r folders. That means that the behavior of Slic3r is different depending on which machine I access. It would be very convenient if Slic3r could have all of it's files self contained in the install directory.

Even if you don't use dropbox or equivalent, another major advantage would be the ability to use various dev and prod versions in parallel with their ini files not interfering with one another, something I have seen cause significant problems with various installations. I find it difficult to try a dev version without stepping on the production version, making it challenging to give feedback on new stuff.

It's possible I just don't know how to configure Slic3r to point to self contained config stuff, but certainly the default behaves as above.

gege2b, 2015-05-09:
did you tried the --datadir option ?
I have exactly the same kind of setup, and putting " --datadir path" did the trick, even with relative path

braddo99, 2015-05-09:
don't know about that, will read up on it!

braddo99, 2015-05-10:
Works great! Now, let's make this the default - is there any reason not to just put the configs in the installation folder? Would be awesome to carry around a flash disk with slic3r and all the configs ready to go.

peter1960, 2015-05-10:
Create a cmd file with the options you need and start it with that.

braddo99, 2015-05-10:
No, it's not problem to add the switch to the windows shortcut and it works great. A cmd would work too. I'm thinking though that there's no reason to put any files outside of the install directory, which would make it possible to run prod and dev builds with no messing around and copying of files.

xoan, 2015-05-10:
Do you know about users? Maybe you have only one user in your system, but trust me, you wouldn't want your ini files to be deleted by your sister.

braddo99, 2015-05-10:
You can always back up your ini files, which I do (in fact dropbox does this for me). However I find it much more important to ensure that I'm using consistent settings across multiple slice machines.

alexrj, 2015-05-13:
I use Dropbox as well, with a symlink (on OS X and Linux).

Putting settings in the app directory is not very standard, and as @xoan said it would break multi-user systems. Also, the application directory might not be writable by all users. It's better to use the standard system paths. Maybe we could make the datadir customizable in the GUI (the path to it would be kept in the system path).

closed 2015-05-13 by alexrj

nicksears, 2015-06-21:
I use dropbox and link shell extension (symbolic link) in windows, but everyone I know wants to be able to customize the profile directory. This is precisely because they have multiple users on one machine, but they all want to use the same profiles (why would you not?) and it's such a pain to keep up with every little change (without symlinks). However, one machine doesn't even work with symlinks because the appdata/roaming/slic3r folder is on a network drive and symlinks doesn't work with network folders. Any update on this? I think it should at least be an option.

nicksears, 2015-06-21:
nvm datadir works great :/ 
for those interested (http://manual.slic3r.org/configuration-organization/configuration-organization)