slic3r/Slic3r #2824: 3d Honeycomb infill affecting perimeters

mattster98, 2015-05-10:
Tried a print with the new 3D Honeycomb infill.  Worked great (as far as I can tell) for the large slab of this model, but the relatively thin vertical pieces had major issues.  It looks like the infill overshot the perimeters, causing gaps.

This is on a well-calibrated printer that cranks out good print after print.  Rostock Max v2.  I usually use rectilinear infill.  I suspect it's caused by resonance/vibration due to the unique infill pattern causing excessive print head wobble, but wanted to see if there are suggestions on how to prevent it, or if it's not an ideal infill for this type of situation.. The gcode looks fine, by the way.  This was 30% infill 3d honeycomb.  0.2 mm layer height

[image]
[image]

I was thinking about trying a test print with a few different options:
1.  Add'l perimeters
2.  Speed/acceleration changes for infill
3.  Different infill%, but I can only imagine higher would make it worse.

braddo99, 2015-05-11:
I haven't tried the 3D honeycomb, but see the same sort of resonance with the regular hexagonal default fill. I have suggested that triangular is not only stronger but also much easier to pull off mechanically. I suspect a tetrahedral fill might be similarly more effective for a triangular 3D approach.

alexrj, 2015-05-17:
I can't see why a machine with no hardware issues would produce that if you say that G-code is fine.

mattster98, 2015-05-18:
I'm sure slowing down the infill speed would solve this.  I fill at this speed with regular honeycomb, rectilinear, and line without issue.

This probably doesn't qualify as a bug, but perhaps more as a feature request.. maybe for this infill type there should be a default infill slow-down factor, or another setting with recommendation.  

Perhaps I'm the only one with this problem - it seems like a cool infill.  I didn't expect it to be a problem.

This layer is what I believe is the first of several layers with problems in the bottom picture of the first post.

[image]

mattster98, 2015-05-19:
OK - printed a couple test pieces using the "oozebane" model as a starting point  Same settings as original piece.  Had to stretch to get the smaller columns to use the 3d honeycomb infill.  Didn't see any issues.  Printed both 0 and 45 degree rotations.  

I'm going to re-print the first piece where this appeared and see if it can be reproduced.  Otherwise I guess this was just a fluke explained by something else.

closed 2015-05-19 by mattster98

mattster98, 2015-05-20:
For anyone interested, I reprinted with the same gcode file, and got similar, but different results.  Still scratching my head as to what exactly is happening, but it's clearly something to do with how my printer is responding.  The infill pattern seems much more square than expected when printed compared to the gcode.
[image]
[image]