slic3r/Slic3r #2825: is nozzle diameter affecting Infill density ?

gege2b, 2015-05-10:
Hi all

I'm not totally sure it's a bug, but it seems that the nozzle diameter affects the generated infill density 

- Load a STL (not model specific)
- Select 20% for infill density and 0.5mm for the nozzle (not really important)
- Look at the 3D preview
- Change the nozzle diameter to a different diameter (say, 0.2mm)
- Look again at the 3D preview

The infill mesh is way more dense

Little nozzle diameter means less extrusion width, that's why I'm not sure it's a bug

Debian 8 x64
current dev

xoan, 2015-05-10:
Infill density is volumetric, so if you configure a thinner extrusion width (it is automatico when you decrease the nozzle diameter), you have a closest infill pattern paths; and if you configure a thicker extrusion width (it is automatic when you increase the nozzle diameter), you have a sparsed infill pattern paths.

alexrj, 2015-05-13:
It's correct.  In order to produce the same density, when a thicker extrusion width is used, larger gaps are generated.

closed 2015-05-13 by alexrj