slic3r/Slic3r #2826: Real extrusion width

rsatue, 2015-05-11:
Hi,

I am looking in detail at the gcode generated by slic3r and Cura. The problem appeared when slicing a file with slic3r resulted in a wrong print and doing the same with Cure was fine. 

When going to the detail of the gcode I saw that the real extrusion width that slic3r makes is much smaller than expected. 

I am calculating the extrusion width in the following way: 

Between two steps, I calculate the volume of material extruded from the difference between the E values time the circumference of the filament, in my case 2,92mm of diameter: Volume = (E2-E1)*(pi*(2,92/2)^2)

Then the value of the width comes from V = distance between points * layer height * width.

I set all the width values to 0,4 to make it easy (all except the default extrusion width) and the result I get is that the first layer is extruded with width = 0.335 mm and the following layers with width = 0.367 mm.

On the other hand, Cura is slicing always with width 0,4mm.

Is this an intended behaviour os slic3r or is it a bug? 

I am using slic3r 1.2.6 and this the result is independent of the stl. 

I've done an exemple with a 100x100x100 cube.
Config file: https://dl.dropboxusercontent.com/u/10708495/Cube%20config.ini
Excel file with the width calculations: https://dl.dropboxusercontent.com/u/10708495/gcode%20analysis.xlsx

Thanks

rsatue, 2015-05-11:
Just one more thing: I have just tried with slic3r 1.1.7 and it makes the width as specified in the advanced settings, in this case width 0,4 mm.

alexrj, 2015-05-11:
No bug at all. Your math is wrong because you assume that extrusion is rectangular, but of course it's not.

See http://manual.slic3r.org/advanced/flow-math

closed 2015-05-11 by alexrj

rsatue, 2015-05-11:
Thanks! My engineering simplification was wrong then! Happy to know that slic3r is taking into account so many details!

There is a joke that says that a mathematician, a physicist and an engineer are trying to measure the volume of a cow. The mathematician says: let's cut the cow in very small parts with known volume, then we add all and we get the volume of the cow. The physicist says: let's submerge the cow into water, we measure how much water it displaces and then we get the volume. And the engineer says: let's suppose an spherical cow...

Obviously I was supposing too much...

alexrj, 2015-05-11:
That's a good one :-)