slic3r/Slic3r #2829: Thin wall generation inconsistent in symetric shape

BurninSun, 2015-05-14:
This is an example of a volcano (for lack of a simpler description - conical frustum with cylindrical hole?) rotated 45 degrees. I don't see any reason that the thin wall generation should be different on one side versus the other, but as can be seen in the picture below, thin walls are mostly generated on the "right" side, with very few on the "left" (along with gaps of none being generated at all).

[image]

Slic3r settings: http://pastebin.com/v3ubs9Fy
STL File: http://pastebin.com/9j6kfvBt
Generated gcode: http://pastebin.com/d8VD385M

alexrj, 2015-05-17:
I will investigate. For self reference, this is one of the layers with a missing thin wall despite the apparent symmetry:

[image]

BurninSun, 2015-05-19:
For the hell of it, I spun these around in different orientations to see if I could narrow the problem down. I tried mirroring and spinning both from within slic3r and with the base stl model file. As long as I ended up in the same orientation, the slicing ended up the same. However, in the process I noticed two things. First, the thin wall setting seems to apply to the infill as well (as visible in your screenshot above - the lower half has those single blue lines for infill while the top half of the model does not). Secondly, the orientation of curvature seems to matter. The following is the same model loaded 4 times and rotated within slic3r software:
[image]
Spinning at 5 degrees at a time also confirmed - the thin walls that get generated are only being generated when their "base" (from the point where 2 lines converge to 1) is at a lower Y value than the next part of the curve. (eg. All of the thin walls in this shot go towards the top of the screen. All of the missing thin walls are ones that would go towards the bottom). This doesn't explain all of the missing thin walls, but it does capture a lot of them.

Note - this doesn't apply to the direction of travel once converted to gcode. On the top left model, left side thin wall gets drawn from the center to its left which, by my description above, would imply it should be missing. But I'm assuming it is generated correctly from the point where the 2 walls converge and going to the right, then simply has its direction of travel inverted when it is converted to gcode for efficiency of the print head travel.

Also to note - if a thin wall starts its curvature by going upward in the Y, it can continue around to travel downward in the Y just fine. It seems to only be the very first segment of thin wall that matters.

alexrj, 2015-05-19:
Your troubleshooting is very good and very correct. I appreciate it. I think the bug is in the math used to compare angles, and I hope to fix it quickly.
(Regarding gap fill, yes, the thin wall setting applies to that as well since it's the same algorithm.)

xoan, 2015-05-19:
Maybe related: #2485

alexrj, 2015-05-21:
Fixed! It was harder than I thought but now it looks much more robust and several other minor issues are now fixed.

[image]

[image]

closed 2015-05-21 by alexrj

BurninSun, 2015-05-22:
Awesome! Way better. Fixed the multitude of random issues I've had and I would suspect many other people's issues as well.

Thank you very much for your work. It is appreciated.

curiouspl2, 2015-05-25:
awesome. noticed this aswell, but did not know it could be improved! prints come out much better now and many details which were trimmed down before are now printed!
thank you !