slic3r/Slic3r #2830: "Only infill where needed" not respecting 0% infill rates

BurninSun, 2015-05-14:
Setting fill density to 0% disables the "Only infill where needed" setting, however that setting is still being taken into account for slicing.

To reproduce:
Set fill density to anything > 0%.
Check "Only infill where needed"
Set fill density to 0%

"Only infill where needed" is now disabled but checked, fill density is 0%, however it results in slicing similar to this:
[image]

Now set fill density to > 0% again.
uncheck "Only infill where needed"
Set fill density back to 0%

"Only infill where needed" is disabled again but unchecked this time, fill density is again 0%, and slicing results in no infill as expected:
[image]

This is happening with the current dev build as well as 1.2.7

Slic3r settings: http://pastebin.com/v3ubs9Fy (other than the "only infill where needed" setting)
STL file: http://pastebin.com/9j6kfvBt

alexrj, 2015-05-18:
Well, that situation is a bit undefined. If you slice that model at ~0.3mm and you ask for top solid shells Slic3r will think that you actually need infill below them, even if you have an internal 0% density:

[image]

I think that slicing a model like that with fill_density = 0 and top_solid_layers = 2 has no valid solutions anyway. No?

BurninSun, 2015-05-18:
I agree that there is not a valid solution with no infill but after setting the infill to 0%, I cannot turn back on the "infill where needed" option because it is disabled. A fix would be one of two changes:

Option 1:
"Only infill where needed" isn't disabled with infill set to 0%. This would let us turn it on if need be, or if sometime in the future we manage to calibrate printers well enough that they can print what is essentially a curved bridge, we would have the option of turning it off.

Option 2:
Force "Only infill where needed" to be on when infill is at 0% so that the print is still valid.

The main problem right now is that the option affects slicing output even though the option is disabled and cannot be changed.

alexrj, 2017-03-18:
*Only infill where needed* is now ignored when Fill Density = 0%.

closed 2017-03-18 by alexrj