slic3r/Slic3r #2831: Walls printing but not infilling

donwallace, 2015-05-14:
I just swapped out hotends and reconfigured slicer. Now when I print walls will print great, but for some reason the walls will not infill between them. Which infill setting handles this?
Thanks you,
 Don

alexrj, 2015-05-17:
See http://manual.slic3r.org/expert-mode/infill

closed 2015-05-17 by alexrj

donwallace, 2015-05-17:
I have already tried everything on that page multiple times and more searching on the net. I am not saying its a bug but what I am doing is asking your expertise on how to fix the issue I am having, not just a response giving me a link to manual, for which I have been through multiple times. With all my search on the net, a ton of people have the same issue. If its under 2mm wall it doesn't infill it, none of which posted a fixed. If this is just the way it is please let me know so I atleast know there is nothing that can be done to correct it. If there is a fix please send my a link to the actual solution, not a link to the manual.
Thank you for your time in this matter,
   Don

alexrj, 2015-05-17:
You asked "Which infill setting handles this?" and the answer is in the manual.

Please help me understand one thing. When you click on "New issue" here on GitHub you're invited to read the guidelines for posting. Why didn't you do it? And if you did, why didn't you do what's written in there? If you want a solution YOU NEED TO PROVIDE THE REQUIRED INFORMATION. Thanks :-)

https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

donwallace, 2015-05-17:
Well at this point I would like to thank you for your time, but I will figure it out on my own. No need to respond.

alexrj, 2015-05-18:
Asking for solutions without providing the required information about the problem is a bad approach, sorry… it will never work. I recommend to fix your attitude towards a more collaborative mood.