slic3r/Slic3r #2832: Multi-Extruder: custom G-code malformed due to incorrect extruder temperature variable resolution

R3D3-yellow, 2015-05-16:
Setup: Renkforce RF1000 3D-printer, PC is Win 8 64 bit running Repetierhost 1.0.6 and Slic3r 1.1.7, connected by USB to the printer.

The RF1000 has been converted to a dual extruder setup, defined in Slic3r, resulting in individual extruder settings.

Problem: with a 2-extruder setup, in Custom G-Code, whenever I use the variables [first_layer_temperature] or [temperature] in a M104 or M109 command, this resolves in the generated G-Code as a double temperature, for instance "215,220" in case the temperature was set to 215° for the first extruder and 220° for the second. Thus, malformed G-Code is created and cannot be processed, resulting in stopping of the print at that location of G-Code.

In the example given, the code generated turns out like:
[code]
M109 S215,220 T0
[/code]

Expected: as I also specify the extruder to which the M104 or M109 applies - either explicitly by name T0 or T1, or implicitly by using T[previous_extruder] or T[next_extruder], I expect the M104 or M109 to be attributed to the correct extruder and therefore the temperature to resolve correctly to (only) the setting related to the specific extruder (which has been previously defined in Filament Settings belonging to the extruder).

Thus:
[code]
M109 S215 T0
[/code]

(I expect the situation to be similar in setups with >2 extruders.)

whosawhatsis, 2015-05-17:
You need [temperature_0] and [temperature_1].

R3D3-yellow, 2015-05-17:
Indeed I might, but as there is only one Custom Code block for tool changes, I need that block to be agnostic on the current or new extruders, i.e. treat these as variables, so I can't just fill in one or the other. (Unless I can embed variables? such as in [temperature_[previous_extruder]] and [temperature_[next_extruder]] ?? That would be neat...).

closed 2015-05-17 by R3D3-yellow

R3D3-yellow, 2015-05-17:
Issue closed - works with [temperature_[previous_extruder]] and [temperature_[next_extruder]]

alexrj, 2015-05-17:
https://github.com/alexrj/Slic3r/wiki/FAQ#what-placeholders-can-i-use-in-custom-g-code

R3D3-yellow, 2015-05-17:
Yes, thanks, I happened to find this independently (despite advanced age) ;-)