slic3r/Slic3r #2833: Possible to turn the auto repair off?

funbart, 2015-05-16:
Hi, I was wondering if it was possible to turn the auto repair off, when loading a model.
I couldn't find an answer on the RepRap forum, the manual, or here, but I can have used inadequate search terms.

With a specific architectural model the repair function is introducing a few  errors, where Netfabb doesn't see them.

Bart

alexrj, 2015-05-17:
No, it's not possible.
Is Slic3r introducing errors in a valid, manifold, model?

rafsoaken, 2015-05-20:
Hi, I have a STL produced by openSCAD that is fine in CURA, but gets "autofixed" by slic3r (1.2.6 and latest stable).
[image]
[image]
[image]

The openSCAD sourcecode (Version 2015.03-1, on Windows 10 tech preview):
´´´
//number of faces
$fn = 120;
layer_height = .2;
nozzle_dia = .4;

//clearance & tolerances
hole_clearance = nozzle_dia; //difference between shaft and hole radii (double for diameter)
shaft_clearance = 3*layer_height; //difference between shaft height and beam thickness
e = 0.05; //epsilon, part overlap

//beam
w = 15; //width
d = layer_height*8; //thickness
length = 120;
r = w/2; //beam end radius
d_end = w/3; //extra distance pins/holes from end

//hole 
hole_perc = .5;
r_hole = r*hole_perc; //radius 

//shaft of pin
r_shaft = r_hole-hole_clearance; //pin radius
h_shaft = d+shaft_clearance; //pin height

//lock piece (on top of pin)
r_lock1 = r_hole+hole_clearance/2; //locking radius
r_lock2 = r_hole-2*hole_clearance; //lock part top radius

h_lock1 = 1.2*d/4; //height of lower part
h_lock2 = 1.4*3*d/4; //height of upper part
h_lock = h_lock1+h_lock2; //lock piece total height

h_pin = h_shaft+h_lock; //total pin height shaft+lock

//pin flat side
d_flatside = r_hole- 2.0*hole_clearance; //distance of flat sides of pin to its center  

//pin slots
w_slit = r_shaft/2; //slit width
l_slit = r_shaft*5; //slit length

module lock_piece(){ // position centered
    //lock piece
    translate([0,0,h_lock1/2-h_lock/2])
        cylinder(h_lock1, r_shaft, r_lock1, center=true);
    translate([0,0,h_lock/2-h_lock2/2])
        cylinder(h_lock2, r_lock1, r_lock2, center=true);  
}

module pin(){ // position centered
    difference(){
        union(){
            //shaft
            translate([0,0, h_shaft/2 - h_pin/2])
                cylinder(h_shaft, r_shaft, r_shaft, center=true);
            translate([0,0, h_pin/2 - h_lock/2]){
                lock_piece();
            }
        }
        
        //cut center slit
        cube([l_slit, w_slit, h_pin+e], center=true);
        
        //cut flat sides
        for(i=[-1,1]) {
            translate([i*(d_flatside+r_lock1/2), 0, 0]) {
                cube([r_lock1, 2*r_lock1+e, h_pin+e], center=true);
            }
        }
    }
}

module slits() {
    //cut center slit
    for(i=[-1,0,1]) {
        translate([0, i*(r_shaft+w_slit/2+e), 0]) { //needs the e or else slic3r loads an incorrect STL, CURA works though
            cube([l_slit*(1-abs(i)*.35), w_slit, d+e], center=true);
        }
    }
}

module hole(){
    cylinder(d*10, r_hole, r_hole, center=true);
}

module beam(){
    difference(){
        hull(){
            cylinder(d, r, r, center=true);
            translate([length-w,0,0]) cylinder(d, r, r, center=true);
        }
        if(length > 79){
        for(i=[1,3]) {
            translate([i*(length-w-2*d_end)/4+d_end,0,0]){
                minkowski(){
                    cylinder([d*2, w/4, w/4]);
                    cube([length/4-l_slit,w/2.9,d*2], center=true);
                }
            }
        }}
    }   
}

difference(){ union(){ //for test prints cutoff
    //beam type 1, 2 holes 1 pin
    difference(){
        beam();
        translate([d_end,0,0]) hole();
        translate([(length-w)/2,0,0]) hole();
        translate([length-w-d_end,0,0]) slits();
    }
    translate([length-w-d_end,0, h_pin/2]) pin();

    //beam type 2, 1 hole, 2 pins 
    translate([0, -1.2*w,0]){
        //beam
        difference(){
            union(){
                beam();
                translate([(length-w)/2,0, h_pin/2]) pin();
                translate([(length-w)-d_end ,0, h_pin/2]) pin();
            }
            translate([d_end,0,0]) hole();
            translate([(length-w)/2 ,0,0]) slits();
            translate([(length-w)-d_end ,0,0]) slits(); 
        }
    }   
}
//translate([2.5*w, -length/1.5, -20]) cube([length,length,100]);
}
´´´

lordofhyphens, 2015-05-20:
As a workaround, try the AMF export in OpenSCAD. Also, you can't be really
sure that Cura will actually give you what you want. I would be certain
that there is actually nothing wrong with the STL (because there have been
bugs in OpenSCAD's STL export before).

You could also send the STL through the online netfabb fixer tool, just to
be sure.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

funbart, 2015-05-21:
Thanks for the suggestions. The stl is an architectural model from ArchiCad, made to test several slicers. I checked it in Netfabb, no problems there, but afterwards I loaded it in Kisslicer as well, and it flagged errors. So it's definitely not a Slic3r problem. It's funny to see though, how the different slicers are solving such errors. 

If wanted I can place a link here, after I finished my test of 5 slicers for a typical architectural model.
(Slic3R, CraftWare, Kisslicer, Cura, Simplify3D)

Thanks!
Bart

closed 2015-05-21 by funbart