slic3r/Slic3r #2834: Unreasonable pressure advance K constant values suggested

hakimio, 2015-05-17:
The tooltip shown when you hover over "Pressure advance" setting states "<..> Reasonable values are in range 0-10". However, if you actually do some testing you'll see that values above 0.02 are quite unreasonable. I think if you care about user experience you should recommend much lower values or just don't recommend any values at all.
There is a section in Sailfish firmware documentation which shows the effects of K constant: http://www.sailfishfirmware.com/doc/tuning-jkn-advance.html . Maybe something similar could be added to slic3r documentation.

alexrj, 2015-05-17:
The advance algorithm in Sailfish firmware is probably implemented in a different way (there's a bit of debate about the formula, and different implementations out there), so their K constant is not the same as Slic3r's K constant.

If you use a value of 2 in Slic3r it will generate a pressure advance of +1.73mm which looks quite reasonable to me.

Can you elaborate more on your testing? Also please specify the Slic3r version you're using, as I think the recent ones have a bug in pressure advance and I don't remember whether I fixed it in 1.2.6 or 1.2.7-dev.

hakimio, 2015-05-17:
I am using 1.2.6. Can you link to the bug you are speaking of?
I have bowden setup with around 55cm bowden tube. I am using 1.75mm filament and e3d v6 hotend. My printer is Prusa i3 with Rambo board and some pretty old Marlin firmware (haven't had any issues with it).
I tried to test "Pressure advance" feature by setting it to 10, 1 and then some values between 0.01 and 0.2. Here is what happened:
* 10 resulted in movement stalling and extruder just pushing plastic in a single spot.
* 1 resulted in no extrusion have of the time.
* Values between 0.01 and 0.1 seemed to noticeable improve the print (corners became noticeably sharper and the print as a whole looked much better). 0.2 resulted in similar result plus I could see that the filling is getting sparse.

hakimio, 2015-05-24:
Maybe you can link to git commit if you can't find relevant issue?

hakimio, 2015-05-26:
I don't see anything related to pressure advance in 1.2.7 changelog and it still says "0-10" are reasonable values while they are clearly not...

Sungod3000, 2015-06-14:
I can somewhat concur. My value is at .75-1 at the moment

@hakimio what speed and acceleration were you using for your tests?

hakimio, 2015-06-14:
@Sungod3000 
I am printing at pretty low speed as my printer is not as sturdy as I would like it to be and I am mostly printing small objects. Here is a screenshot of my config:
[image]

Actually I gave up on pressure advance setting as even with such a low value as 0.1 it would cause to randomly start extruding large amounts of plastic in a singe spot (unfortunately I no longer have the generated gcode). I think it's poorly tested, there is no documentation and @alexrj doesn't care about it that much as he is using direct drive and has more important issues to resolve and features to implement. Of course, a bit more input from @alexrj than just putting a couple of tags on the issue would be nice.

Also, I guess @llluis who made [the original patch](https://github.com/alexrj/Slic3r/pull/2018) might be a bit too busy to take a look at the issue too.

Sungod3000, 2015-06-14:
Aha, my print speeds arent much higher then yours around 50mms and 500 acceleration. You're right indeed there isnt much documentation. I searched for some formulas on how to estimate a good k- value but came up short. I think It also has to be coupled directly to speed and acceleration. 
I see you have acceleration disable therefore your need for pressure advance you be lower right? or do you see fringed corners?

Im planning another testing battery once I have my frame stabilized and get some higher acceleration settings. 

Another interesting thing is how to k value (and the acceleration setting as well) relates to the new autospeed feature. I like it very much, but does that mean that the deceleration is disabled in autospeed mode since I would be a speed change?

hakimio, 2015-06-14:
Yeah, one of the main reasons I started experimenting with pressure advance was that when I moved from direct to bowden I could see that in some places corners were much sharper on prints done with direct drive. For me even a small pressure advance value makes a big difference, but I haven't experimented a lot with it. Only tried 0.01-0.2, 1 and 10.

To be honest, I haven't played with acceleration in slic3r, so pretty much all acceleration values are set to defaults. What kind of acceleration settings are you using and how does it effect your prints?

Where did you find "autospeed"? I am using slic3r with Repetier Host, so I might be missing some settings.

lordofhyphens, 2015-06-15:
Autospeed is brand new to 1.2.8.

I bet repetier doesn't have a gui option for that yet.

Just use the stock UI :p and you will find it under print settings -> speed

Sungod3000, 2015-06-15:
Acceleration I am at the moment on a conservative 500-1000mm/s^2 mainly to not shake my printer apart and reduce acceleration ripples. I still some of them but they are much reduced compared to the 3000-5000 that marlin has hardcoded.

CCS86, 2015-07-31:
I am just testing out pressure advance on my heavily modified UMO tonight.

I started at 5, and while parts of the print looked good, it seemed to be leaving some patches completely unfilled. I am printing at 2 now and it seems better. I will try 0.1 as well.

I'm also trying firmware retraction, but despite having enabled it in my firmware, I do not see the different retract/advance speeds during the print.

hakimio, 2015-08-19:
As long as the problem described by CapnBry in issue #2960 is not solved, pressure advance can't be used.

ioquatix, 2016-05-27:
Wondering where we are at with this?

VanessaE, 2016-06-07:
Once built, it crashes when run:
`No adaptive_slicing in ConfigOptionsGroup config at /usr/share/perl5/Slic3r/GUI/OptionsGroup.pm line 344.`

If I move my Slic3r config out of reach, I can run it as long as I keep it in "Simple" mode.  Switch it to "Expert" mode and exit/restart, and it crashes with the same error.

Here's the full log of what I did (git output, build log, etc, 1532 lines):
http://pastebin.ubuntu.com/17074218/

VanessaE, 2016-06-07:
Please ignore that last message from me.  Posted on the wrong issue.

lordofhyphens, 2016-06-07:
@ioquatix Still of the opinion that it's better done in the firmware myself. As autospeed is supposed to minimize retracts for most printing (by keeping the pressure constant) I wouldn't try using the two together.

l29ah, 2018-02-01:
Do i get it right that "pressure advance" == "velocity extrusion"?

l29ah, 2020-04-16:
Today at pressure advance = 2 slic3r rewinds a few centimeters of filament, then seemingly tries to print an external perimeter, then pulls it back in, then proceeds with other perimeters. gcode: 
[Pedal_Strap_upper.txt](https://github.com/slic3r/Slic3r/files/4489147/Pedal_Strap_upper.txt)

aaaidan, 2021-05-30:
I agree with @l29ah. 

Pressure Advance looks like it has worked in the past, but doesn't seem helpful as of Slic3r 1.3.0. I also can't find documentation or commentary for it (official or otherwise). 

Since there wasn't much online about it, I thought I'd dive in to see if I could get it working. I made a small hollow T-shape to test various pressure advance values, to print on an Ender 3 V2.
[image]

* I printed a control (pressure advance 0). It printed great, other than the expected small bulges at the corners, from deceleration.
* I tried a value of 2. Extruder made angry noises and the print failed. 😢  Possibly pushing too much filament without moving at the start. 
* With a value of 0.5, the brim printed a bit funny (blob at start), and the starting corner of the model had a smaller blob. **However, all other corners look identical to the control print.**

Looking at the gcode, Slic3r seems to offset the extruder only a few times at the start and end of the print, but doesn't attempt any other adjustment for places with nozzle acceleration, like corners. The model has eight right angles over 36 layers, so I was expecting the gcode to be dotted with hundreds of pressure adjustments.

[image]

I may be missing something necessary to get it to work correctly. Anyone using this parameter at the moment?

Unless there are plans to resurrect it, **perhaps it's worth considering removing, or at least deprecating, Pressure Advance**. At best, it's confusing and gives false hope 😆 

While it would be an awesome feature for slic3r to have, there are alternatives available in Marlin (for machines that support its Linear Advance) and Klipper's Pressure Advance.