slic3r/Slic3r #2835: 1.2.7-dev: parts jumping out of sight in 3D plater when moving around

simonkuehling, 2015-05-18:
Hi,

i've found a glitch in the 3D plater: when moving around several parts, sometimes the part that was just moved jumps far out of sight, away from the cursor - but can be found again when zooming out quite a bit.

This could possibly overlap with #2728 - anyway, i made a screencast where i just managed to reproduce this effect:

[image]

kefir-, 2015-05-18:
I've experienced this recently as well, I think it was in 1.2.7-dev. I had created multiple versions of an item, and I think I had some other parts plated as well. I was manually arranging pieces when all the parts suddenly disappeared off the bed like in the animation.

lordofhyphens, 2017-03-19:
I haven't noticed this with 1.3.0-dev. 

Please comment if you can get it to reproduce I'll reopen.

closed 2017-03-19 by lordofhyphens